ros2/rviz #363: rviz2 MarkerArray Display error

sagniknitr, 2018-10-16:
Selecting the MarkerArray Display in the latest rviz2 is showing `No transform to fixed frame [base_link]. TF error: [Lookup would require extrapolation into the past. Requested time 1511969984.89534 but the earliest data is at time 1511970003.02050, when looking up transform from frame [odom] to frame [base_link]]` error in the display window.

The said MarkerArray message contains markers with two different frame_id : 1) base_link and 2)odom.

All the markers with odom frame_id are not displayed in the rviz2 window.

[image]

sagniknitr, 2018-10-22:
A update on this issue : the exception seems to come from the function  `TFFrameTransformer::transform(const rviz_common::transformation::PoseStamped & pose_in, const std::string & target_frame)   ` in the file. (https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/src/rviz_default_plugins/transformation/tf_frame_transformer.cpp)

It seems that using `ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 map odom ` is the only solution to the issue. I don't understand why rviz2 needs to listen to `/tf_static` topic.  published by the static_transform_publisher,  to update the TF tree.

Martin-Idel-SI, 2018-10-23:
Yes, definitely. The error message is a tf error as the message tells you, hence must come from the `lookupTransform` call in the display, which calls `transform`. In other words, it must come from within `TFFrameTransformer`.

Static transforms are no problem, because their transform is available at any time as the name suggests.

This is probably yet another issue that will be solved once message filters are available - or at least the real problem should become much more apparent!

sagniknitr, 2018-10-23:
Thanks at follow up @Martin-Idel-SI . I thought it may be related to tf2 message_filters, but needed some confirmation.

shiveshkhaitan, 2019-02-08:
@sagniknitr if the error still persists can you please tell how to reproduce the same. I have an implementation of rviz tf message filters ready [PR](https://github.com/ros2/rviz/pull/375). It would be helpful in confirming if that actually serves the purpose.

closed 2020-05-29 by sagniknitr