ros2/rviz #364: rviz_rendering doesn't link assimp

paoloach, 2018-11-02:
Hi,
using gentoo distribution, I installed ROS2 from the source but I had some problem compiling rviz_rendering because it doesn't link the assimp library.

I had to modify the CMakeLists.txt file in order to add the library.
target_link_libraries(rviz_rendering
   rviz_ogre_vendor::OgreMain
   rviz_ogre_vendor::OgreOverlay
**+  assimp**
   Qt5::Widgets
   ament_index_cpp::ament_index_cpp
   resource_retriever::resource_retriever

The commit id is
42e3c4b47bc6acd90ed690b884bcc385555c382c

Paolo

wjwwood, 2018-11-02:
This should not be required as the linking against `assimp` should be done by `rviz_assimp_vendor`, see:

https://github.com/ros2/rviz/blob/42e3c4b47bc6acd90ed690b884bcc385555c382c/rviz_rendering/CMakeLists.txt#L124-L126

> The commit id is 42e3c4b

What does that commit have to do with `assimp` or `rviz_rendering`?

paoloach, 2018-11-03:
Hi wjwwood,

Let me explain why I open this bug. As written, I installed ROS2 from source in my **gentoo** distribution. 
When it was the turn of rviz_rendering to be compiling, I got a lot of linking error of type undefined reference to `Assimp::xxxxxx .
Adding the assimp library this errors dissapeared.

But, as you state, rviz_rendering depends on rviz_assimp_vendor and so cmake should understand to use assimp.
So, it can be a cmake version problem considering that gentoo use a different version of it ?
I don't know if it worth to investigate futher considering I don't use the ufficial distribution.

P.S.
The commit should be use to understand the source code it refers the bug: I used the HEAD, but tomorrow the HEAD can point to a different commit, so declaring the commit to which the bug refers fix in the time the source code.

allenh1, 2018-11-08:
> So, it can be a cmake version problem considering that gentoo use a different version of it ?

It's probably a Gentoo CMake patch that's at fault here, actually. I'm still working on getting Gentoo to build colcon and friends correctly (still failling on notifications, but nothing critical to the build process).

Not sure if you are familiar with the [ROS-Overlay](https://github.com/ros/ros-overlay), but I'd be willing to try and work through this there with you so that we don't make too much noise in the ROS 2 repos. I suspect this is an issue with my patches for colcon, and the ordering of the paths got flipped or something.

paoloach, 2018-11-09:
Honestly, I know ROS very little. 
With few modification of the same type as this, I was able to compile ros2 with colcon.
The greatest change I did is patch the  python setuptools because it try to install the  python packets into .../lib while install/...../colcon use the python system library that in my case install/....../lib64

But I stopped  to this result because my real target was to compile on Raspberry and so I switched to a docker machine for running ROS2. 

But it will be fine for me work together to build and run ROS2 on Gentoo.
Let me know how to proceed. 

Paolo

clalancette, 2023-07-14:
Since this is so old, I'm going to close this out.  If you are still having problems, please feel free to open a new issue and give us details on what operating system you are using, and the exact steps you took.  Thanks.

closed 2023-07-14 by clalancette