ros2/rviz #368: Flaky tests on Linux-aarch64 build

jacobperron, 2018-12-21:
E.g. https://ci.ros2.org/job/ci_linux-aarch64/2467/

Seems that repeating tests is causing something to crash and not report test results.

Looks like this issue was introduced by #297.

wjwwood, 2018-12-21:
Dang, I guess the CI was old, but I figured that nothing else had really changed on rviz... I'll revert it and look into the issue.

wjwwood, 2018-12-21:
Reverted in https://github.com/ros2/rviz/pull/369

cottsay, 2018-12-27:
Is there anything else to be done here, or did #369 sole the issue?

wjwwood, 2019-01-02:
I was leaving this open until the revert could be undone, but I guess we could close this if desired since it only really applies to fixing the CI, not also fixing the originating change and getting it re-merged.

Martin-Idel, 2019-01-04:
The revert will definitely not have changed anything. The commit only introduced changes in `rviz_default_plugins`, while the test failures start with tests in `rviz_rendering` (e.g. `point_cloud_renderable_test_target`, which is built before `rviz_default_plugins`.

I recall that we had the same test failures in the nightlies when enabling display tests on Linux systems back in July and I don't remember them ever having been resolved. I tried to debug the issue once but couldn't reproduce - but it seems to have been Ogre not coming up successfully...

jacobperron, 2019-01-07:
There has been a slew of test failures in the nightlies over the past few days for `rviz_default_plugins` and `rviz_rendering`: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/655/
The symptoms are missing test results.
@Martin-Idel Sounds like the same issues you experienced in the past?

wjwwood, 2019-01-07:
I was working under an assumption that @jacobperron saw these as new flaky tests, indicating this was true:

> Looks like this issue was introduced by #297.

If the flaky-ness was preexisting then we could undo the revert and address the flaky tests separately. I didn't investigate deeply. I was just trying to keep CI clean leading up to the release.

andreasholzner, 2019-01-08:
I am sure that the flaky aarch64 tests are not related to #297 as we have seen those also in the past.  We had no direct access to aarch64 machines and thus could not debug as these problems do not show on amd64.

wjwwood, 2019-01-14:
So it seems like the changes in https://github.com/ros2/rviz/pull/297 were not the cause of this? Does it just make it more flaky? I'm about to do the release for Crystal patch 1 and need to resolve this...

jacobperron, 2019-01-14:
> So it seems like the changes in #297 were not the cause of this? Does it just make it more flaky? I'm about to do the release for Crystal patch 1 and need to resolve this

It does seem that they are not the cause. I wasn't aware of the previous failures when I originally reported. I can't say that it makes it more flakey. I think I opened this issue after seeing the failures for two or three days. I'd say it would be okay to add the change back.

jacobperron, 2019-01-14:
Here's the (original?) issue I missed: ros2/build_cop#144

andreasholzner, 2019-01-18:
I could (almost) reproduce the test results, however, the console output is a little different.
##### Output seen while trying to reproduce
> Running main() from gmock_main.cc
> [==========] Running 3 tests from 1 test case.
> [----------] Global test environment set-up.
> [----------] 3 tests from PointCloudRenderableTestFixture
> [rviz_rendering:debug] Available Renderers(1): OpenGL Rendering Subsystem, at /home/ubuntu/rviz_ws/src/rviz/rviz_rendering/src/rviz_rendering/render_system.cpp:273

The line starting with `[rviz_rendering:debug]` is not present in the failing ci builds. In both cases the tests fail with a segfault.

I could trace the segfault to the call to `Ogre::Root::createRenderWindow`. https://github.com/ros2/rviz/blob/069eb10f1153569c6281f1f6efb7e16a59c7fe95/rviz_rendering/src/rviz_rendering/render_system.cpp#L529

The call stack is not helpful.
> (gdb) bt
> #0  0x0000ffffbc77e670 in ?? () from /usr/lib/aarch64-linux-gnu/libGLX.so.0
> Backtrace stopped: previous frame identical to this frame (corrupt stack?)

I am out of ideas. Maybe a Ogre upgrade could help, but this would require some work. During a quick try to use the Ogre 1.11.5 the `rviz_ogre_vendor` package did not compile.

dirk-thomas, 2019-03-14:
Is there any chance that these test failure will be addressed in the near future? If not I would propose to exclude the affected packages (`rviz_common`, `rviz_default_plugins`, `rviz_rendering`, `rviz_rendering_tests`) from the nightly aarch64 repeated job.

andreasholzner, 2019-03-15:
I don't see how so I am in favor of excluding the packages from the nightly job.
After an Ogre upgrade they could be enabled again.

Martin-Idel, 2019-10-17:
This should have been solved by #394 (at least that's what my test showed there). Are the tests reenabled? Could somebody try it again?

clalancette, 2021-05-03:
I just ran a repeated job with this stuff reenabled: https://ci.ros2.org/job/ci_linux-aarch64/9301/.  While there are test failures, none of them are in rviz-related components, so this is probably fixed.  I'm going to close this out and open a PR to reenable these tests on the nightlies.

closed 2021-05-03 by clalancette