ros2/rviz #377: rviz2 composable node?

lucasw, 2019-02-21:
Would it be as simple as putting the contents of  [rviz2 main.cpp](/ros2/rviz/blob/ros2/rviz2/src/main.cpp) into a Node, routing arguments into the qapp and vapp, and adding CLASS_LOADER_REGISTER_CLASS and so on?

wjwwood, 2019-02-21:
No, not quite, as that main function creates a node itself... I'm not sure what they right way to organize this would be. Probably a Node that does the Qt setup which is in the main function, as you suggested, but that also passes itself (as `this`) into the rviz code so that it uses that one and does not create another node internally. Also, this new node that does the Qt setup and so on, would maybe also need to inherit from the node class in rviz (if there's a custom one, I don't remember offhand).

wjwwood, 2019-02-21:
I'm going to go ahead and close this, as it's kind of a question. Feel free to continue discussion and if needed I can reopen it.

However, I think the right thing would be to open a new feature request issue and/or maybe smaller issues that outline what needs to be changed to achieve this. Sorry I don't have more time to think about it or prototype something.

closed 2019-02-21 by wjwwood