ros2/rviz #383: glXCreateContextAttribsARB() function not found

ubald, 2019-03-13:
Hi,
I just started trying our ROS2 and I can't launch rviz2. I am running an Ubuntu 18.04 guest in Parallels on Mac. I searched the issues for something similar and found this: https://github.com/ros2/rviz/issues/338 which was merged into ogre in 2017.

So anyway here's the log I get when starting it with `-l`:

```
$ rviz2 -l
[INFO] [rviz2]: Creating resource group General
[INFO] [rviz2]: Creating resource group OgreInternal
[INFO] [rviz2]: Creating resource group OgreAutodetect
[INFO] [rviz2]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[INFO] [rviz2]: Registering ResourceManager for type Material
[INFO] [rviz2]: Registering ResourceManager for type Mesh
[INFO] [rviz2]: Registering ResourceManager for type Skeleton
[INFO] [rviz2]: MovableObjectFactory for type 'ParticleSystem' registered.
[INFO] [rviz2]: ArchiveFactory for archive type FileSystem registered.
[INFO] [rviz2]: DDS codec registering
[INFO] [rviz2]: ETC codec registering
[INFO] [rviz2]: stb_image - v2.15 - public domain JPEG/PNG reader
[INFO] [rviz2]: Supported formats: jpeg,jpg,png,bmp,psd,tga,gif,pic,ppm,pgm,hdr
[INFO] [rviz2]: Registering ResourceManager for type HighLevelGpuProgram
[INFO] [rviz2]: Registering ResourceManager for type Compositor
[INFO] [rviz2]: MovableObjectFactory for type 'Entity' registered.
[INFO] [rviz2]: MovableObjectFactory for type 'Light' registered.
[INFO] [rviz2]: MovableObjectFactory for type 'BillboardSet' registered.
[INFO] [rviz2]: MovableObjectFactory for type 'ManualObject' registered.
[INFO] [rviz2]: MovableObjectFactory for type 'BillboardChain' registered.
[INFO] [rviz2]: MovableObjectFactory for type 'RibbonTrail' registered.
[INFO] [rviz2]: *-*-* OGRE Initialising
[INFO] [rviz2]: *-*-* Version 1.10.11 (Xalafu)
[INFO] [rviz2]: OverlayElementFactory for type Panel registered.
[INFO] [rviz2]: OverlayElementFactory for type BorderPanel registered.
[INFO] [rviz2]: OverlayElementFactory for type TextArea registered.
[INFO] [rviz2]: Registering ResourceManager for type Font
[INFO] [rviz2]: Loading library /opt/ros/crystal/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL
[INFO] [rviz2]: Installing plugin: GL RenderSystem
[INFO] [rviz2]: OpenGL Rendering Subsystem created.
[INFO] [rviz2]: Plugin successfully installed
[INFO] [rviz2]: CPU Identifier & Features
[INFO] [rviz2]: -------------------------
[INFO] [rviz2]:  *   CPU ID: GenuineIntel: Intel(R) Xeon(R) CPU E5-1650 v2 @ 3.50GHz
[INFO] [rviz2]:  *          SSE: yes
[INFO] [rviz2]:  *         SSE2: yes
[INFO] [rviz2]:  *         SSE3: yes
[INFO] [rviz2]:  *        SSE41: yes
[INFO] [rviz2]:  *        SSE42: yes
[INFO] [rviz2]:  *          MMX: yes
[INFO] [rviz2]:  *       MMXEXT: yes
[INFO] [rviz2]:  *        3DNOW: no
[INFO] [rviz2]:  *     3DNOWEXT: no
[INFO] [rviz2]:  *         CMOV: yes
[INFO] [rviz2]:  *          TSC: yes
[INFO] [rviz2]:  *INVARIANT TSC: yes
[INFO] [rviz2]:  *          FPU: yes
[INFO] [rviz2]:  *          PRO: yes
[INFO] [rviz2]:  *           HT: no
[INFO] [rviz2]: -------------------------
[INFO] [rviz2]: ******************************
*** Starting GLX Subsystem ***
******************************
[INFO] [rviz2]: GLX_VERSION = 1.4
[INFO] [rviz2]: GLX_EXTENSIONS = GLX_ARB_get_proc_address GLX_EXT_texture_from_pixmap GLX_SGIX_fbconfig
[INFO] [rviz2]: Registering ResourceManager for type Texture
[INFO] [rviz2]: GLRenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed  miscParams: FSAA=4 border=none contentScalingFactor=1.000000 currentGLContext=false externalWindowHandle=92274689 left=0 parentWindowHandle=92274689 top=0 
[INFO] [rviz2]: Actual frame buffer FSAA: 0, gamma: false
[ERROR] [rviz2]: RuntimeAssertionException: glXCreateContextAttribsARB() function not found in createNewContext at /tmp/binarydeb/ros-crystal-rviz-ogre-vendor-5.1.0/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/RenderSystems/GLSupport/src/GLX/OgreGLXGLSupport.cpp (line 859)
[ERROR] [rviz2]: rviz::RenderSystem: error creating render window: RuntimeAssertionException: glXCreateContextAttribsARB() function not found in createNewContext at /tmp/binarydeb/ros-crystal-rviz-ogre-vendor-5.1.0/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/RenderSystems/GLSupport/src/GLX/OgreGLXGLSupport.cpp (line 859)
[INFO] [rviz2]: GLRenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed  miscParams: FSAA=4 border=none contentScalingFactor=1.000000 currentGLContext=false externalWindowHandle=92274689 left=0 parentWindowHandle=92274689 top=0

... 100x more ...

[ERROR] [rviz2]: Unable to create the rendering window after 100 tries
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
Aborted (core dumped)
```

I also tried building from the source in case it had been fixed but the result is the same unfortunately.

Thanks in advance for your help!

Martin-Idel, 2019-03-17:
There are issues with RViz in virtualization and at some point it didn't work on Mac and was never fixed: https://github.com/ros2/rviz/issues/338). 

The trouble should be the supported OpenGL version (maybe you can see your Linux guest's OpenGL version?). RViz currently uses the OpenGL 3+ Plugin which needs at least OpenGL 3.2 (if I recall correctly). The error says that some OpenGL functions are missing which normally means that the necessary OpenGL version is not (fully?) supported.

wjwwood, 2019-03-18:
The error comes from within ogre itself (they get a pointer to an API call and never check it for nullptr), but even if they checked, they'd have to do something different if that API is not available.

I fixed at least one case of this here: https://github.com/OGRECave/ogre/pull/507

But there are more cases, and that only gives you a better error code.

I was hoping a future version of Ogre would address this issue, but I don't that's the case yet.

ubald, 2019-03-20:
I switched to VMWare and I was able to run Rviz2 since VMWare Fusion supports OpenGL 3. Unfortunately I had to go back to ROS1 for other unrelated reasons. Thanks for your help! Should this issue be kept open if it's already a known issue?