ros2/rviz #385: rviz2 segmentation error in OS X

vmayoral, 2019-03-17:
### Description

rviz2 fails to launch in OS X Mojave 10.14

### Your environment
* ROS Distro: Crystal patch 2
* OS Version: OS X Mojave 10.14
* Source or Binary build? source
* If source, which git commit or tag? 5.0.0

### Behaviour
```bash
| => rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[INFO] [rviz2]: Stereo is NOT SUPPORTED
2019-03-17 15:25:36.175 rviz2[13893:5135322] -[OS_dispatch_mach frame]: unrecognized selector sent to instance 0x7fdaebafb630
Illegal instruction: 4
```

### Backtrace or Console output

https://gist.github.com/vmayoral/7eababb372c320f1243c3c4eeb2b7d88

vmayoral, 2019-03-17:
For the record, I'm currently trying out the same with ROS 2 Crystal Clemmys - Patch Release 3 compiled from sources.

vmayoral, 2019-03-17:
Connects with https://github.com/ros2/ros2/issues/671

vmayoral, 2019-03-17:
Fails also with patch 3 release: https://gist.github.com/vmayoral/856524030d27931aee985a3a4ecb9433

vmayoral, 2019-03-17:
Interesting thing is that it throws different errors on consecutive launches:
```bash
| victor:~/ros2_ws$
| => rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[INFO] [rviz2]: Stereo is NOT SUPPORTED
2019-03-17 19:19:50.387 rviz2[89080:5619313] -[__NSCFType frame]: unrecognized selector sent to instance 0x7f86ca80a7c0
Illegal instruction: 4
________________________________________________________________________________
| victor:~/ros2_ws$
| => rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[INFO] [rviz2]: Stereo is NOT SUPPORTED
2019-03-17 19:19:58.742 rviz2[89084:5619825] -[_NSXPCConnectionClassCache frame]: unrecognized selector sent to instance 0x7fb748aaffe0
Illegal instruction: 4
________________________________________________________________________________
| victor:~/ros2_ws$
| => rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[WARN] [rviz2]: Could not find folder /Users/victor/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[INFO] [rviz2]: Stereo is NOT SUPPORTED
Segmentation fault: 11
```

vmayoral, 2019-03-17:
Similar results obtained while installing things from binaries: https://gist.github.com/vmayoral/947dfcc0afacd0552d4b313bfa34def4

wjwwood, 2019-03-18:
It wasn't built for Mojave, so it doesn't surprise me that it doesn't work.

I also currently do not have it working on Mojave. I don't believe I'll have any time to investigate right now.

wjwwood, 2019-03-18:
Btw, our CI/binaries are for Sierra, though I regularly used it with High Sierra for a long time. See:

https://index.ros.org/doc/ros2/Installation/OSX-Install-Binary/#id1

vmayoral, 2019-03-18:
> I also currently do not have it working on Mojave. I don't believe I'll have any time to investigate right now.

Totally understood. I'll look into downgrading for now and will try to book some time and look into this myself.

Karsten1987, 2019-05-20:
@vmayoral It looks like the changes proposed in [here](https://github.com/ros2/rviz/pull/380#issuecomment-493152628) work for Mojave, do you mind giving that a try?

vmayoral, 2019-05-20:
@Karsten1987, sure thing. From a quick look, seems like there's been some updates in `ogre`. Though that might do so I first tried quickly to update `ogre`:

```bash
brew install ogre
Updating Homebrew...
Ignoring path homebrew-cask/
To restore the stashed changes to /usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask run:
  'cd /usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask && git stash pop'
==> Auto-updated Homebrew!
Updated 3 taps (osrf/simulation, homebrew/core and homebrew/cask).
==> New Formulae
atlantis                                                                                  libzt
==> Updated Formulae
...
==> Installing ogre from osrf/simulation
Error: Your Xcode (10.1) is too outdated.
Please update to Xcode 10.2.1 (or delete it).
Xcode can be updated from the App Store.
```
Updated XCode to the latest version and re-launched it:

```bash
_____________________________________________
| victor:~/ros2_ws/src/ros2/geometry2 (ros2)$
| => brew install ogre
Updating Homebrew...
Ignoring path homebrew-cask/
To restore the stashed changes to /usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask run:
  'cd /usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask && git stash pop'
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/cask).
No changes to formulae.

==> Installing ogre from osrf/simulation
Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences.

Error: Cannot install osrf/simulation/ogre because conflicting formulae are installed.
  ogre1.9: because Differing version of the same formula

Please `brew unlink ogre1.9` before continuing.

Unlinking removes a formula's symlinks from /usr/local. You can
link the formula again after the install finishes. You can --force this
install, but the build may fail or cause obscure side effects in the
resulting software.
________________________________________________________________________________
| victor:~/ros2_ws/src/ros2/geometry2 (ros2)$
| => brew unlink ogre1.9
Unlinking /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8... 27 symlinks removed
________________________________________________________________________________
| victor:~/ros2_ws/src/ros2/geometry2 (ros2)$
| => brew install ogre
==> Installing ogre from osrf/simulation
Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Preferences.

==> Installing dependencies for osrf/simulation/ogre: boost, cmake and tbb
==> Installing osrf/simulation/ogre dependency: boost
==> Downloading https://homebrew.bintray.com/bottles/boost-1.69.0_2.mojave.bottle.tar.gz
==> Downloading from https://akamai.bintray.com/80/8089ad2fdc0edffcd0222043fda9a99bf82abf30f334305b9068fbef85b44893?__gda__=exp=1558386247~hmac=4f0d9befeaabf6cd20ae9dd99d8794075c
######################################################################## 100.0%
==> Pouring boost-1.69.0_2.mojave.bottle.tar.gz
🍺  /usr/local/Cellar/boost/1.69.0_2: 13,717 files, 489.4MB
==> Installing osrf/simulation/ogre dependency: cmake
==> Downloading https://homebrew.bintray.com/bottles/cmake-3.14.4.mojave.bottle.tar.gz
==> Downloading from https://akamai.bintray.com/59/5966170978a7df7552799372ebe2b440e0f1b413a318f87707f50a7d3063a2a5?__gda__=exp=1558386290~hmac=3028bf7b337851aef096426c2136ea6a3d
######################################################################## 100.0%
==> Pouring cmake-3.14.4.mojave.bottle.tar.gz
==> Caveats
[... 70 lines cut ...]
patching file CMake/Dependencies.cmake
patching file RenderSystems/GL/CMakeLists.txt
patch unexpectedly ends in middle of line
Hunk #1 succeeded at 218 with fuzz 1.
==> Applying ogre-78331c592397878ab3377b5f9b55086aca7a63e5.diff
patching file CMakeLists.txt
==> cmake -DCMAKE_OSX_ARCHITECTURES='x86_64' -F/Library/Frameworks/ -lfreetype -DOGRE_BUILD_PLUGIN_CG=OFF -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCM
==> make install
🍺  /usr/local/Cellar/ogre/1.7.4: 493 files, 15.3MB, built in 4 minutes
==> Caveats
==> cmake
Emacs Lisp files have been installed to:
  /usr/local/share/emacs/site-lisp/cmake
```
tried launching again:
```bash
victor:~/ros2_ws$
| => rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
Segmentation fault: 11
```

That didn't do so I'll fetch the latest sources and compile again the `ros2_ws`. Will report back tomorrow morning when it finishes compiling.

wjwwood, 2019-05-21:
rviz in ROS 2 has it's own copy of Ogre, so the one in Homebrew shouldn't affect it. Though I haven't tried that lately.

vmayoral, 2019-05-21:
@Karsten1987, fyi, I'm facing some issues in Mojave with Qt after updating. Particularly:
```bash
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp
CMake Error at /usr/local/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:15 (message):
  The imported target "Qt5::Core" references the file

     "/usr/local/.//mkspecs/macx-clang"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/Qt5Core/Qt5CoreConfigExtras.cmake:50 (_qt5_Core_check_file_exists)
  /usr/local/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:169 (include)
  /usr/local/lib/cmake/Qt5Gui/Qt5GuiConfig.cmake:83 (find_package)
  /usr/local/lib/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:83 (find_package)
  /usr/local/lib/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  src/qt_gui_cpp/CMakeLists.txt:3 (find_package)
```

AFAIK, this not a known issue. I'll explore this in more detail tonight and report back.

vmayoral, 2019-05-22:
Still stuck with Qt. I can't put more time on this for now. Let's see if I can review it again over the weekend.

vmayoral, 2019-05-22:
All right, managed to reproduce it finally. This issue still applies to me:

```bash
rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
Segmentation fault: 11
```

the dump https://gist.github.com/vmayoral/1955bc258f94425e9c9c3923d7bcc05b

vmayoral, 2019-05-26:
Doing https://github.com/ros2/ros2/issues/700#issuecomment-496026773 my issue finally got fixed. There seems to be an issue with the default `ogre` from brew and osrf's fixes it.

Closing here and thanks for the support.

closed 2019-05-26 by vmayoral