ros2/rviz #395: Update crystal deb packages - Request

nzlz, 2019-04-24:
Would it be possible to update the crystal branch and release a new version of the debian packages?

With the crystal packages the URDF visualization via /robot_description topic is not working, while with latest sources it does. It would be nice if we could rely on a crystal deb package while the dashing version arrives.

sloretz, 2019-04-24:
@nzlz do you know which PR fixed the issue? If you can find it, and it can be backported without breaking ABI, then it can be added to the next crystal patch release https://github.com/ros2/ros2/issues/684

nzlz, 2019-04-24:
Seem like the commit that solved the issue was: https://github.com/ros2/rviz/commit/023dddcea431395cff9f0a830c17f1fb19941976, coming from PR https://github.com/ros2/rviz/pull/378 .

sloretz, 2019-04-24:
@wjwwood do you think #378 can be backported? The removal 1 and addition of 2 virtual methods would break ABI in a publicly accessible class, but since the class is a `Display` plugin does that kind of ABI change matter?

https://github.com/ros2/rviz/blob/314150c4c38d9ef4a56b0dc1750ea8f1c928683c/rviz_default_plugins/include/rviz_default_plugins/displays/robot_model/robot_model_display.hpp#L118-L119

sloretz, 2019-04-24:
@nzlz it looks like Dashing will be the first distro to have that fix. The PR was dropped from Crystal patch release 3 (ros2/ros2#656) because of the API change.

```
Display the robot description correctly in rviz. Breaks API. Will not be released into Crystal.

    ros2/rviz#382 backports:
        ros2/rviz#378
```

In the meantime you might have to build `rviz_default_plugins` from source. https://index.ros.org/doc/ros2/Tutorials/Colcon-Tutorial/

nzlz, 2019-04-24:
Thanks @sloretz ! we will have to wait a few weeks then.

closed 2019-04-24 by nzlz