ros2/rviz #400: Map is not rendered if not published before rviz launch

bergercookie, 2019-05-15:
I'm following the tutorial of the nav2 stack: https://github.com/ros-planning/navigation2/tree/master/nav2_bringup

Following its instructions, rviz2 is launched first and subscribes to the `/map` topic which is not published yet. Then I'm launching `amcl` which is publishing map messages to `/map`. I can pick inspect the latter from the command line `ros2 topic echo /map` but not from rviz2 *unless* I restart the latter

mkhansen-intel, 2019-06-20:
@wjwwood - this issue is a result of the /map topic using transient_local, reliable QOS, which I mentioned in the TSC

mkhansen-intel, 2019-07-08:
@wjwwood - can we get an update as to when rviz will support transient local reliable QOS?

wjwwood, 2019-07-11:
I think @jacobperron is going to have a look at it, since he's messing with stuff in the displays classes already, but a community pull request (not necessarily from you @mkhansen-intel or @bergercookie) is always appreciated for things like this.

wjwwood, 2019-07-11:
I think this would a good candidate for backport as well, depending on how the changes look.

mkhansen-intel, 2019-07-22:
@jacobperron - any update on this?

jacobperron, 2019-07-22:
@mkhansen-intel No, I haven't had a chance to take a look yet.

jacobperron, 2019-08-30:
> Checking again, can we get this on the Eloquent roadmap?

I should have some time next week to take a look at this.

jacobperron, 2019-09-07:
I took a look, and the easiest way to resolve this issue is to override the default QoS durability setting in the MapDisplay constructor to "transient local" (see #461). This has the implication that communication won't work with any map publishers that do not have the same durability setting.

As an alternative (or longer term solution), maybe we should consider adding properties for durability and other QoS settings to the topic displays. @wjwwood @Martin-Idel, what do you think?

Martin-Idel, 2019-09-07:
You could also just add a bool property to the map display asking for whether subscription should be transient local or not - similar to the unreliable property. This would allow easy backporting.

If this setting and other QoS settings are relevant for RViz in other places as well, I'd also think that adding them to the topic displays is a cleaner solution (see #409 for a starting point, but I guess that's why I was pinged?). I think my approach in #409 would make the code cleaner and allow a unified user experience even if some settings will hardly ever be used for some displays.

jacobperron, 2019-09-10:
@Martin-Idel I actually overlooked #409. I think that makes sense and we could iterate on that.

jacobperron, 2019-09-20:
I believe this issue is resolved by https://github.com/ros2/rviz/pull/409, which introduces QoS properties for topic displays. You should be able to set the map display topic's *durability* setting to *transient_local* and see maps published before RViz was launched (presuming that the publisher has the same setting).

closed 2019-09-20 by jacobperron

mkhansen-intel, 2019-09-20:
I can't wait to try it!