ros2/rviz #415: resource retriever error for rviz_common

Karsten1987, 2019-06-14:
When trying to build `rviz_visual_testing_framework` I get the following error:
```
CMake Error in CMakeLists.txt:
  Imported target "rviz_common::rviz_common" includes non-existent path

    "/usr/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

Can anybody explain what's happening here? I am building rviz with `--merge-install` for what it's worth and repeating the colcon build command seems to resolve the problem.

Martin-Idel, 2019-06-19:
Just to be sure: You're not using an overlay workspace? 

If not, I find it very peculiar.

If you are, this might be an artifact of the use of target names within `target_link_libraries` instead of `ament_target_dependencies`, especially since you mention the resource retriever in the title.

clalancette, 2023-07-14:
Given the age of this, I'm going to close this out.  We fixed a lot of these types of bugs during the Galactic/Humble timeframe, so I think it should be fixed.  But if you are still having the issue please feel free to reopen.

closed 2023-07-14 by clalancette