ros2/rviz #417: When setting a 2D Pose Estimate, RViz incorrectly logs "Setting goal"

rotu, 2019-06-19:
When setting a 2D Pose Estimate, RViz incorrectly logs "Setting goal".
Below is the log output from setting a 2D Pose Estimate then setting a 2D Nav Goal:
```
[INFO] [rviz2]: Setting goal: Frame:map, Position(1.30849, -2.7475, 0), Orientation(0, 0, 0.624316, 0.781172) = Angle: 1.34851
[INFO] [rviz2]: Setting goal: Frame:map, Position(-2.8186, 2.15941, 0), Orientation(0, 0, 0.555852, 0.831281) = Angle: 1.17878
```

testkit, 2019-07-03:
This is a blocker issue for navigation, see https://github.com/ros-planning/navigation2/issues/905 and https://github.com/ros-planning/navigation2/issues/901.

rotu, 2019-07-03:
> This is a blocker issue for navigation

I don't think this issue should block you. Despite that the log says "setting goal", it still publishes an initial pose that I have successfully used in AMCL to seed localization.

testkit, 2019-07-03:
Yeah, I probably should say "the rviz2 no responding issue" is a blocker issue which we met in both ros-planning/navigation2#905 and ros-planning/navigation2#901.

Martin-Idel, 2019-07-03:
But that's a different issue. This issue is about the fact that the log message for the 2d pose estimate is wrong. That's true but it definitely doesn't affect any publishing of the goal itself.

You should probably open a different issue for each potential bug for tracking purposes.

PR #427 (2019-07-17): Disambiguate "estimate" pose from "goal" pose in log
Fixes #417 
Previously setting a 2D Pose Estimate then a 2D Nav Goal would log something like:
```
[INFO] [rviz2]: Setting goal: Frame:map, Position(1.30849, -2.7475, 0), Orientation(0, 0, 0.624316, 0.781172) = Angle: 1.34851
[INFO] [rviz2]: Setting goal: Frame:map, Position(-2.8186, 2.15941, 0), Orientation(0, 0, 0.555852, 0.831281) = Angle: 1.17878
```

Now it logs something like:
```
[INFO] [rviz2]: Setting estimate pose: Frame:map, Position(0.672111, -0.269852, 0), Orientation(0, 0, -0.964214, 0.265125) = Angle: -2.60493
[INFO] [rviz2]:  [... cut]
closed 2019-07-17 by jacobperron