ros2/rviz #424: RViz crash with high-resolution map

rotu, 2019-07-05:
RViz seems to have trouble rendering high-pixel count maps, causing a very reproducible crash.
Publishing the map in `map_lores.zip` (533x423 px, 230.3 kB) and trying to view it in rviz works fine.
Publishing the map in `map_hires.zip` (861x661 px, 569.2 kB) causes a hard crash with the following error info.

```
ros2 launch openrover_demo rviz.launch.py 
[INFO] [launch]: All log files can be found below /home/snuc/.ros/log/2019-07-05-16-22-21-496956-rhea-15882
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [15913]
[rviz2-1] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[rviz2-1] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [rviz2]: Creating 1 swatches_
[rviz2-1] [ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /opt/ros/master/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[rviz2-1] [ERROR] [rviz2]: Failed to create 1 swatches_
[rviz2-1] [INFO] [rviz2]: Creating 2 swatches_
[rviz2-1] [ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /opt/ros/master/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[rviz2-1] [ERROR] [rviz2]: Failed to create 2 swatches_
[rviz2-1] [INFO] [rviz2]: Creating 4 swatches_
[rviz2-1] [ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /opt/ros/master/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[rviz2-1] [ERROR] [rviz2]: Failed to create 4 swatches_
[rviz2-1] [INFO] [rviz2]: Creating 8 swatches_
[rviz2-1] [ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /opt/ros/master/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[rviz2-1] [ERROR] [rviz2]: Failed to create 8 swatches_
[rviz2-1] [ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /opt/ros/master/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[rviz2-1] terminate called after throwing an instance of 'Ogre::InvalidParametersException'
[rviz2-1]   what():  InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /opt/ros/master/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2-1]: process has died [pid 15913, exit code -6, cmd '/opt/ros/master/install/lib/rviz2/rviz2 --display-config /home/snuc/ros2_ws/src/openrover/openrover_demo/config/default.rviz --fixed-frame map'].
```

[map_hires.zip](https://github.com/ros2/rviz/files/3364037/map_hires.zip)
[map_lores.zip](https://github.com/ros2/rviz/files/3364040/map_lores.zip)

rotu, 2019-07-05:
It seems to be related to the total image size, not the maximum dimension, as cropping either the width or height to 400px prevents the crash.

Martin-Idel, 2019-07-06:
This is a non-obvious duplicate of #394.

The root cause is that any swatch can only have at most 255*255 pixels and there can only be eight swatches in total (this is the same in the rviz code - I wonder why it doesn't provide a problem there).

I'm investigating whether there is a better fix than just raising the number of swatches.

rotu, 2019-07-06:
@Martin-Idel awesome. I’m glad you understand this issue! Will try the PR.

rotu, 2019-07-07:
Upon further reflection and research, I think (1) this has nothing to do with #394. (2) this is due to an errant `static_cast<uint8_t>` in `Swatch::resetTexture` accidentally introduced in #267.

ROS2:
https://github.com/ros2/rviz/blob/ros2/rviz_default_plugins/src/rviz_default_plugins/displays/map/swatch.cpp#L212
ROS1:
https://github.com/ros-visualization/rviz/blob/4b6c0f447159044bfaa633e140e6094b19516b02/src/rviz/default_plugin/map_display.cpp#L204

A swatch *should be able* to be more than 255 bits wide and high, but these dimensions are incorrectly truncated to a `uint8_t` instead of a `uint16_t` (aka `ushort`, for those of you following along in the code). This effectively limits each swatch to 255 bits in each dimension - it should be able to be bigger.

Martin-Idel, 2019-07-07:
That's also partly my fault - I first thought it'd be exactly the same issue and more swatches will indeed solve the problem. But as you say I then thought that a swatch should be able to hold more data and found the root cause.

I edited my post above but left in the part about being a "non-obvious duplicate". Sorry for that confusion!

PR #425 (2019-07-30): Fix map display
Fixes #424
Fixes #401 

This was a bug introduced during migration:
The Ogre API allows textures of size at most `unsigned short` number of bytes height or width. However, the current implementation only creates textures of at most 8 bit height or width. Since the number of swatches was capped at 8 in the original RViz and this limit has been kept, this means that maps of moderate size (e.g. 800 * 800 pixels) crashed. 

This PR fixes this bug by creating textures of 16 bit size.

I also improved the status quo:
- If a map is too large, a message will be logged and the map will  [... cut]
closed 2019-07-30 by jacobperron