ros2/rviz #437: rviz crashes when loading navigation2 config with opensplice or RTI connext as the dds

Jconn, 2019-08-01:
I am running on ubuntu 18.04 ros2 dashing binary release

I am trying to run the navigation2 stack with opensplice as my dds implementation, but when I load rviz with the config file with 

ros2 run rviz2 rviz2 -d nav2_bringup/launch/nav2_default_view.rviz

rviz crashes

```
[johnconn@amadeus navigation2 (dashing-devel ✗)]$ ros2 run rviz2 rviz2 -d nav2_bringup/launch/nav2_default_view.rviz
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[INFO] [rviz2]: Stereo is NOT SUPPORTED

```

I originally posted this issue here: https://github.com/ros-planning/navigation2/issues/996

and there is a similar issue with using RTI connext posted on navigation2 here: https://github.com/ros-planning/navigation2/issues/916

crdelsey, 2019-08-01:
I am able to reproduce the crash by just adding the map display to the default Rviz config and then subscribing to the map topic. I documented the procedure in #438 before I realized @Jconn filed this issue. Thanks @Jconn

SteveMacenski, 2019-08-01:
@crdelsey @Jconn it look like @jacobperron fixed that about a day ago https://github.com/ros2/rviz/commit/4b3fcb427302b3f0ff449e980d92c12311d926ae

pull in 6.1.3 and see if that fixes your problem.

Jconn, 2019-08-01:
I'm getting the same behavior with opensplice

jacobperron, 2019-08-02:
I can reproduce with the 6.1.3 on top of dashing release debs.
When I build all dashing branches from source, the problem seems to be resolved. So, I think this will be fixed with the next Dashing patch release (https://github.com/ros2/ros2/issues/736).

Jconn, 2019-08-04:
@jacobperron building from the 'master' vcs script is working for me, thanks you. I am comfortable closing this at your discretion.

jacobperron, 2019-08-05:
We can re-open this if you experience issues following the next Dashing patch release.

mkhansen-intel, 2019-08-07:
I'm testing this today and seeing it with latest debians, assuming those are yesterdays release as I installed today. I can reproduce with the same steps @crdelsey used in #438:
```
source /opt/ros/dashing/setup
export RMW_IMPLEMENTATION=rmw_opensplice_cpp
rm -rf ~/.rviz2 (just to ensure you start with the default config)
Add in the map display
Set the topic to /map
```

jacobperron, 2019-08-08:
This is the error I see:

```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create subscription: type support handle implementation 'rosidl_typesupport_fastrtps_cpp' (0x7fbe3e6b6f08) does not match valid type supports ('rosidl_typesupport_opensplice_cpp' (0x7fbed2d60098), 'rosidl_typesupport_opensplice_c' (0x7fbed2f625c8)), at /tmp/binarydeb/ros-dashing-rmw-opensplice-cpp-0.7.3/src/rmw_subscription.cpp:100, at /tmp/binarydeb/ros-dashing-rcl-0.7.6/src/rcl/subscription.c:167
```

jacobperron, 2019-08-08:
I do not have this issue building the [tagged release](https://github.com/ros2/ros2/blob/6486fbd0b97e88e338c1ffc2de26156614d6117a/ros2.repos) from source, so it appears there's something funny happening during packaging.

jacobperron, 2019-08-08:
Looks like the issue is that type support for OpenSplice and Connext are not being shipped with the debians for `maps_msgs` (which is used by the map display):

```
$ ls /opt/ros/dashing/lib/libmap_msgs__*                                                                        
/opt/ros/dashing/lib/libmap_msgs__python.so                  /opt/ros/dashing/lib/libmap_msgs__rosidl_typesupport_fastrtps_cpp.so
/opt/ros/dashing/lib/libmap_msgs__rosidl_generator_c.so      /opt/ros/dashing/lib/libmap_msgs__rosidl_typesupport_fastrtps_c.so
/opt/ros/dashing/lib/libmap_msgs__rosidl_typesupport_cpp.so  /opt/ros/dashing/lib/libmap_msgs__rosidl_typesupport_introspection_cpp.so
/opt/ros/dashing/lib/libmap_msgs__rosidl_typesupport_c.so    /opt/ros/dashing/lib/libmap_msgs__rosidl_typesupport_introspection_c.so
```

nuclearsandwich, 2019-08-08:
Packages which provide ROS interfaces should be declared as members of the `rosidl_interface_packages` group in order to make sure that bloom injects the build dependency on all type supports: https://github.com/ros-infrastructure/bloom/blob/07d8f1c8783989beb4c6f2f5aeae4388f2b38f80/bloom/generators/rosdebian.py#L104-L121

So this can be resolved with a PR and a re-release of `maps_msgs` as we move toward community interface packages this is something we should either revisit or broadcast more widely. I've put a task in my queue to audit the interface packages released via the ROS 2 build farm for this group declaration.

jacobperron, 2019-08-08:
But map_msgs is already declared as part of `rosidl_interface_packages`: https://github.com/ros-planning/navigation_msgs/blob/ab82ad25398d97d77fc91433e4e1c62d4acca920/map_msgs/package.xml#L35 :thinking:

jacobperron, 2019-08-08:
@dirk-thomas Helpfully pointed out that `map_msgs` was last released with bloom 0.7.2, which didn't have necessary fixes (https://github.com/ros-infrastructure/bloom/pull/502) to add the `rosidl_interface_packages` dependencies for Dashing. I've re-released `map_msgs`, which will hopefully fix this issue after the next sync: https://github.com/ros-infrastructure/bloom/pull/502

nuclearsandwich, 2019-08-09:
> But map_msgs is already declared as part of `rosidl_interface_packages`: https://github.com/ros-planning/navigation_msgs/blob/ab82ad25398d97d77fc91433e4e1c62d4acca920/map_msgs/package.xml#L35 thinking

Ah yeah, I was looking at the wrong branch.

mkhansen-intel, 2019-08-12:
Building navigation_msgs from source (ros2 branch) then running Rviz with Opensplice works. So we can do that until the next Dashing sync.

dirk-thomas, 2019-08-12:
You can also use the Debian packages available from the testing repo before they are synced to the main repo.

mkhansen-intel, 2019-08-12:
I'm not sure how to do that, are there instructions somewhere?

dirk-thomas, 2019-08-12:
> I'm not sure how to do that, are there instructions somewhere?

E.g. https://discourse.ros.org/t/ros-2-dashing-diademata-call-for-testing-and-package-releases/8819

Jconn, 2019-09-19:
rviz is now loading the navigation2 config for me with dashing release packages - I am closing this. Thank you for your help everyone.

closed 2019-09-19 by Jconn