ros2/rviz #439: RVIZ hangs on 6.1.2/6.1.3

SteveMacenski, 2019-08-01:
Hi,

I launch default `ros2 run rviz2 rviz2` and it just hangs. Eventually when I control+c I get

```
Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.
```

jacobperron, 2019-08-01:
Can you clarify what exactly you built?

I tried building rviz `6.1.3` on top of dashing debs, but am unable to reproduce.

SteveMacenski, 2019-08-01:
I built a clean workspace of just ros2 [repo file](https://github.com/ros2/ros2/blob/master/ros2.repos) packages, then I see this issue with no additional changes.

jacobperron, 2019-08-01:
> I built a clean workspace of just ros2 repo file packages, then I see this issue with no additional changes.

Of the [master branches](https://github.com/ros2/ros2/blob/master/ros2.repos) or [dashing branches](https://github.com/ros2/ros2/blob/dashing/ros2.repos)? I wouldn't expect RViz 6.1.X to work with the master branches.

SteveMacenski, 2019-08-01:
On master, got it, thanks!

closed 2019-08-01 by SteveMacenski