ros2/rviz #440: [osx][rviz_rendering] using isless undefined

Karsten1987, 2019-08-05:
I have this problems when trying to build RViz on a overlay workspace on my OSX machine:

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:321:9: error: no member named 'isless' in the global namespace
using ::isless;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:322:9: error: no member named 'islessequal' in the global namespace
using ::islessequal;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:323:9: error: no member named 'islessgreater' in the global namespace
using ::islessgreater;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:324:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:325:9: error: no member named 'isunordered' in the global namespace
using ::isunordered;
      ~~^
13 errors generated.
make[2]: *** [CMakeFiles/rviz_rendering.dir/src/rviz_rendering/ogre_render_window_impl.cpp.o] Error 1
make[1]: *** [CMakeFiles/rviz_rendering.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_rendering	[ Exited with code 2 ]
```

I googled quite a bit around and the closest I've come to this issue that it's somewhat related to whether one includes `math.h` before `cmath` but I can't really make sense out of it.
https://github.com/Andersbakken/rtags/issues/937

Any ideas? I am pretty much stuck with it.

scpeters, 2019-08-05:
I've seen this a fair bit in osx on high sierra (10.13), but I think you're using mojave. On high sierra, it seemed to happen when using a newer macOS SDK than the host system. We change some environment variables to work around it:

* https://github.com/osrf/homebrew-simulation/blob/master/Formula/ignition-msgs1.rb#L61-L62

but I'm not sure if this is the same problem that you are having

Karsten1987, 2019-08-05:
Looking into it mit `VERBOSE=1 make` yields the following:
```
[  6%] Automatic MOC for target rviz_rendering
/usr/local/Cellar/cmake/3.15.1/bin/cmake -E cmake_autogen /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/CMakeFiles/rviz_rendering_autogen.dir/AutogenInfo.cmake Debug
[  6%] Built target rviz_rendering_autogen
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/rviz_rendering.dir/build.make CMakeFiles/rviz_rendering.dir/depend
cd /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering && /usr/local/Cellar/cmake/3.15.1/bin/cmake -E cmake_depends "Unix Makefiles" /Users/karsten/workspace/osrf/ros2_robot/src/rviz/rviz_rendering /Users/karsten/workspace/osrf/ros2_robot/src/rviz/rviz_rendering /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/CMakeFiles/rviz_rendering.dir/DependInfo.cmake --color=
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/rviz_rendering.dir/build.make CMakeFiles/rviz_rendering.dir/build
[  7%] Building CXX object CMakeFiles/rviz_rendering.dir/rviz_rendering_autogen/mocs_compilation.cpp.o
/usr/local/opt/ccache/libexec/c++  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_WIDGETS_LIB -DRVIZ_RENDERING_BUILDING_LIBRARY -Drviz_rendering_EXPORTS -I/Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering -I/Users/karsten/workspace/osrf/ros2_robot/src/rviz/rviz_rendering -I/Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/rviz_rendering_autogen/include -I/Users/karsten/workspace/osrf/ros2_robot/src/rviz/rviz_rendering/include -I/usr/local/include/eigen3 -I/usr/local/include -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/OSX -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/Bites -isystem /usr/local/include/SDL2 -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/HLMS -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/MeshLodGenerator -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/Overlay -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/Paging -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/Property -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/RTShaderSystem -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/Terrain -isystem /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/Volume -iframework /usr/local/opt/qt5/lib -isystem /usr/local/opt/qt5/lib/QtWidgets.framework/Headers -isystem /usr/local/opt/qt5/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers -isystem /usr/local/opt/qt5/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt5/./mkspecs/macx-clang -isystem /Users/karsten/workspace/osrf/ros2_full/install/include -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include  -Werror -Wno-c++11-extensions -Wno-unknown-warning-option -g -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk -fPIC   -Wall -Wextra -Wpedantic -std=c++14 -fPIC -std=gnu++14 -o CMakeFiles/rviz_rendering.dir/rviz_rendering_autogen/mocs_compilation.cpp.o -c /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/rviz_rendering_autogen/mocs_compilation.cpp
In file included from /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/rviz_rendering_autogen/mocs_compilation.cpp:2:
In file included from /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/rviz_rendering_autogen/YRS6ZQGVSV/moc_render_window.cpp:10:
In file included from /Users/karsten/workspace/osrf/ros2_robot/build/rviz_rendering/rviz_rendering_autogen/YRS6ZQGVSV/../../../../src/rviz/rviz_rendering/include/rviz_rendering/render_window.hpp:39:
In file included from /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/OgreSceneNode.h:31:
In file included from /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/OgrePrerequisites.h:376:
In file included from /Users/karsten/workspace/osrf/ros2_robot/install/opt/rviz_ogre_vendor/include/OGRE/OgreStdHeaders.h:28:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:313:9: error: no member named 'signbit' in the global namespace
using ::signbit;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:314:9: error: no member named 'fpclassify' in the global namespace
using ::fpclassify;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cmath:315:9: error: no member named 'isfinite' in the global namespace; did you mean 'finite'?
using ::isfinite;
      ~~^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/math.h:749:12: note: 'finite' declared here
extern int finite(double)
```

I am on Mojave 10.14.6 and thus it should be okay with respect to version SDK.

However, I managed to get it built with disabling the tests for this repo.
```
colcon build --cmake-args -DBUILD_TESTING=OFF
```
I hope this limits the search space.

oneattosecond, 2019-08-20:
Hello, I was referred here from a failure to build nav2_rviz_plugins - see https://github.com/ros-planning/navigation2/issues/1043

Can anyone kindly suggest how to modify the rviz_ogre_vendor package to not include the MacOS SDK root as a system include? Or if I'm barking up the wrong tree, I can file a separate issue

oneattosecond, 2019-08-20:
some relevant sounding discussion over here: https://gitlab.kitware.com/cmake/cmake/issues/19180

Karsten1987, 2019-08-21:
Can you give the latest RViz `ros2` branch a shot? We just merged an upgrade to Ogre 1.12.1, which hopefully resolves the problem. It did it for me.

https://github.com/ros2/rviz/pull/394

oneattosecond, 2019-08-22:
@Karsten1987 Do you have a .repos file handy? I tried building stock Dashing ros.repos + grabbing latest ros2 branch of Rviz, and rviz_common failed to build due to missing tf2/buffer_core_interface.h, would prefer to not guess on dependencies.

jacobperron, 2019-08-22:
@oneattosecond Try using the latest repos file: https://github.com/ros2/ros2/blob/master/ros2.repos

oneattosecond, 2019-08-23:
dang, I'm getting a bunch of link errors in nav2_rviz_plugins... at least it gets farther???

--- stderr: nav2_rviz_plugins
Undefined symbols for architecture x86_64:
  "rviz_default_plugins::tools::PoseTool::deactivate()", referenced from:
      vtable for nav2_rviz_plugins::GoalTool in mocs_compilation.cpp.o
  "rviz_default_plugins::tools::PoseTool::onInitialize()", referenced from:
      nav2_rviz_plugins::GoalTool::onInitialize() in goal_tool.cpp.o
  "rviz_default_plugins::tools::PoseTool::processMouseEvent(rviz_common::ViewportMouseEvent&)", referenced from:
      vtable for nav2_rviz_plugins::GoalTool in mocs_compilation.cpp.o
  "rviz_default_plugins::tools::PoseTool::activate()", referenced from:
      vtable for nav2_rviz_plugins::GoalTool in mocs_compilation.cpp.o
  "rviz_default_plugins::tools::PoseTool::PoseTool()", referenced from:
      nav2_rviz_plugins::GoalTool::GoalTool() in goal_tool.cpp.o
  "rviz_default_plugins::tools::PoseTool::~PoseTool()", referenced from:
      nav2_rviz_plugins::GoalTool::GoalTool() in goal_tool.cpp.o
      nav2_rviz_plugins::GoalTool::~GoalTool() in goal_tool.cpp.o
  "typeinfo for rviz_default_plugins::tools::PoseTool", referenced from:
      typeinfo for nav2_rviz_plugins::GoalTool in mocs_compilation.cpp.o
ld: symbol(s) not found for architecture x86_64

wjwwood, 2019-10-08:
> Can you give the latest RViz ros2 branch a shot? We just merged an upgrade to Ogre 1.12.1, which hopefully resolves the problem. It did it for me.
> 
> #394

@Karsten1987 did this resolve the problem for you? I'm building on master and have the same issue.

Karsten1987, 2019-10-08:
It did re-appear on my machine on a fresh build. My workaround to build it is by disabling the tests:

```
colcon build --cmake-args -DBUILD_TESTING=OFF
```

I know that's not a real solution, but at least it makes RViz compile and workable.

wjwwood, 2019-10-25:
@mjcarroll did you say you investigated this and thought it was a test that wasn't being built on CI due to the lack of a display?

nuclearsandwich, 2019-10-25:
> @mjcarroll did you say you investigated this and thought it was a test that wasn't being built on CI due to the lack of a display?

That was the theory. I was waiting to test it before commenting and got mired in qt installation grief. Having worked through it, rviz_rendering appears to build albeit with some stderr. I get the same results on our Mojave host lore when building from a GUI terminal (accessed via VNC) as the buildfarm yields when building in a "headless" terminal during the CI job.

Example from https://ci.ros2.org/job/nightly_osx_release/1429/

```
--- stderr: rviz_rendering
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libgmock_main.a(mocs_compilation.cpp.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libgmock_main.a(mocs_compilation.cpp.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libgmock.a(mocs_compilation.cpp.o) has no symbols
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: libgmock.a(mocs_compilation.cpp.o) has no symbols
---
```

Host details:
* MacOS 14.4.4 Build 18E226
* XCode 10.2.1 Build 10E1001

paroj, 2020-03-13:
are you building with -Werror? https://gitlab.kitware.com/cmake/cmake/issues/19845