ros2/rviz #442: rviz2 crashes when launched as a node

rotu, 2019-08-07:
When launching using a `launch_ros.actions.Node`, RViz2 crashes, choking on the newly introduced `--ros-args` option. Seems to be caused by https://github.com/ros2/launch_ros/pull/52

```
from pathlib import Path

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import SetEnvironmentVariable, DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node

def generate_launch_description():
    rviz_config = Path(get_package_share_directory('openrover_demo'), 'config', 'default.rviz').resolve()
    assert rviz_config.is_file()

    return LaunchDescription([
        DeclareLaunchArgument('frame', default_value='map', description='The fixed frame to be used in RViz'),
        SetEnvironmentVariable('RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED', '1'),
        Node(
            package='rviz2', node_executable='rviz2', output='screen',
            arguments=[
                '--display-config', str(rviz_config),
                '--fixed-frame', LaunchConfiguration(variable_name='frame')
            ]
        )
    ])
```

```
$ ros2 launch openrover_demo rviz.launch.py
[INFO] [launch]: All log files can be found below /home/dan/.ros/log/2019-08-07-14-17-41-837727-rhea-5588
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [6352]
[rviz2-1] args before = /home/dan/ros2_ws/install/lib/rviz2/rviz2 --display-config /home/dan/ros2_ws/install/share/openrover_demo/config/default.rviz --fixed-frame map --ros-args 
[rviz2-1] args before = /home/dan/ros2_ws/install/lib/rviz2/rviz2 --display-config /home/dan/ros2_ws/install/share/openrover_demo/config/default.rviz --fixed-frame map --ros-args 
[rviz2-1] Unknown option 'ros-args'.
[ERROR] [rviz2-1]: process has died [pid 6352, exit code 1, cmd '/home/dan/ros2_ws/install/lib/rviz2/rviz2 --display-config /home/dan/ros2_ws/install/share/openrover_demo/config/default.rviz --fixed-frame map --ros-args'].
```

mikaelarguedas, 2019-08-07:
:+1: I'm facing the same issue

rotu, 2019-08-07:
It worked for me to rewrite my launch file as below. I'm not sure if rviz *should* be launched as a node, though that seems to be a common practice.

```
from pathlib import Path

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import SetEnvironmentVariable, DeclareLaunchArgument, ExecuteProcess
from launch.substitutions import LaunchConfiguration

def generate_launch_description():
    rviz_config = Path(get_package_share_directory('openrover_demo'), 'config', 'default.rviz').resolve()
    assert rviz_config.is_file()

    return LaunchDescription([
        DeclareLaunchArgument('frame', default_value='map', description='The fixed frame to be used in RViz'),
        SetEnvironmentVariable('RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED', '1'),
        ExecuteProcess(cmd=[
                'rviz2',
                '--display-config', str(rviz_config),
                '--fixed-frame', LaunchConfiguration(variable_name='frame')
            ],
            output='screen'),
    ])
```

jacobperron, 2019-08-22:
I ran into a similar issue where I attempted to use the new option as follows:

```
rviz2 --ros-args ...
```

or

```
ros2 run rviz2 rviz2 --ros-args ...
```

I guess this has something to do with rviz being run as a Qt application.

orduno, 2019-09-03:
Additionally, prior to [ros2/launch_ros#52](https://github.com/ros2/launch_ros/pull/52), if `rviz` was launched as a process the following worked for setting a namespace and remapping some topic:

```python
    start_rviz_remapped_cmd = launch.actions.ExecuteProcess(
        cmd=[os.path.join(get_package_prefix('rviz2'), 'lib/rviz2/rviz2'),
            ['__ns:=/', rviz_namespace],
            ['/some_topic:=some_topic_remapped'],
        cwd=[launch_dir], output='screen')
```

This no longer works, the namespace is not applied and some_topic is not remapped.

orduno, 2019-09-03:
The parsing is done during the call to `init` of `VisualizerApp`:

https://github.com/ros2/rviz/blob/c28f2c2798b6c3bc21486bf2e0572952e3eb875a/rviz_common/src/rviz_common/visualizer_app.cpp#L129

(Seems like) all that is needed is to have it ignore this flag by adding:

```cpp
  QCommandLineOption ros_args(
    QStringList() << "r" << "ros-args",
    "ROS arguments passed to rclcpp::init");
  parser.addOption(ros_args);
```

probably here:

https://github.com/ros2/rviz/blob/c28f2c2798b6c3bc21486bf2e0572952e3eb875a/rviz_common/src/rviz_common/visualizer_app.cpp#L178

The arguments are still passed to `rcl`

https://github.com/ros2/rviz/blob/c28f2c2798b6c3bc21486bf2e0572952e3eb875a/rviz_common/src/rviz_common/visualizer_app.cpp#L280

https://github.com/ros2/rviz/blob/c28f2c2798b6c3bc21486bf2e0572952e3eb875a/rviz_common/src/rviz_common/ros_integration/ros_client_abstraction.cpp#L60

I tested both, launching as `Node` or `ExecuteProcess` with `ros_args`

orduno, 2019-09-06:
Also choking on the -r/--remap option introduced in https://github.com/ros2/launch_ros/pull/59

lbegani, 2019-09-09:
As a temporary solution, checked out https://github.com/ros2/launch_ros/commit/606eaee60db693c22c1afff4f5dd6c794084e81e to proceed.

mkhansen-intel, 2019-10-17:
@jacobperron @wjwwood - there are work-arounds for this like launching rviz as a proess instead of node but that means every launch file that is currently doing that will need to change in order to work with Eloquent. Can we get this on the Eloquent must-fix list please?

jacobperron, 2019-10-17:
@mkhansen-intel I'll look into resolving this.

jacobperron, 2019-10-22:
Please see https://github.com/ros2/rviz/pull/474 for a proposed fix.

closed 2019-10-22 by jacobperron

mkhansen-intel, 2019-10-22:
Thanks @jacobperron!

PR #474 (2019-10-22): Remove ROS arguments before passing to QApplication
Resolves #442
commit 8e68df4c (2019-10-22): Remove ROS arguments before passing to QApplication (#474)

Resolves #442