ros2/rviz #443: Improve error messages from TF message filter

jacobperron, 2019-08-08:
Steps to reproduce:

1. Launch the dummy robot

```
ros2 launch dummy_robot_bringup dummy_robot_bringup.launch.py
```

2. Start RViz

```
rviz2
```

3. Changed fixed frame to `odom`

4. Add a display for an existing topic, e.g. LaserScan on with topic `/scan`

Output:

The following message repeated:

```
[INFO] [rviz]: [signalFailure] Drop message: frame 'single_rrbot_hokuyo_link' at time 1565205069.594 for reason(0)
```

Expected output:

It would be nice to see the actual reason for the message being dropped.

We could either improve the error in the TF message filter itself, or provide a way for RViz to access the error code and log the error with the node logger.

rotu, 2019-08-08:
see https://github.com/ros2/geometry2/pull/130

anderwm, 2021-01-08:
Any advice how to get to the root of this problem?

[INFO] [1610126663.962665070] [rviz]: Message Filter dropping message: frame 'rgb_camera_link' at time 1610126658.085 for reason 'Unknown'

closed 2021-01-13 by ivanpauno