ros2/rviz #444: [tool plugins] "2D Nav Goal" -> "Goal Pose" & topic update

SteveMacenski, 2019-08-09:
Over in navigation2 we're talking about the topic to use for the topic-based navigation. Right now we use the `move_base_simple/goal` topic, but that's obviously not really awesome considering move_base is gone.  https://github.com/ros-planning/navigation2/issues/1021

We propose changing that topic to `goal_pose` and updating the button name appropriately. I think that makes it good for navigation as well as more "generalized" for non-nav tasks. 

If you guys are OK with that change, I'll submit a PR here for those updates.

jacobperron, 2019-08-22:
Makes sense to me. I'm happy to review a PR.

SteveMacenski, 2019-08-22:
:+1: Good to ask before going through the motions. I'll do today or tomorrow.

closed 2019-09-04 by SteveMacenski