ros2/rviz #458: Rviz2 crash when loading camera laser scan.

MengNan-Li, 2019-09-02:
I  conver depth image  to laser scan by the  depthimage_to_laserscan package. Rviz2 crash when loading camera laser scan.

[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: Creating 1 swatches_
[ERROR] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : active samplers with a different type refer to the same texture image unit
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
Aborted (core dumped)

gbalke, 2021-09-22:
Hi @MengNan-Li, I noticed the error had an error related to storing a negative value as a time. Can I ask for a sample image/camera_info pair that you attempted to send that causes this error?

I wasn't able to create the issue until I set the `header.stamp.sec` to -1:

```cpp
depth_msg_.header.frame_id = "map";
depth_msg_.header.stamp.sec = -1;  
```

```
[INFO] [1632349832.427020105] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1632349832.427109032] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1632349832.442226727] [rviz2]: Stereo is NOT SUPPORTED
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
Aborted (core dumped)
```

This shouldn't crash RViz so I'll be making a fix for the time-related error but I don't believe there's anything inherently wrong with depthimage_to_laserscan as it does not generate this negative number.

gbalke, 2021-10-14:
I'm closing this as could not reproduce and haven't had a response for some time. If this comes up again please re-open.