ros2/rviz #463: Vertex shader error when rendering with MapDisplay

jacobperron, 2019-09-09:
To reproduce, add a map display and publish a map to render.
When the first map message is processed, we see the following error message:

```
[ERROR] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
active samplers with a different type refer to the same texture image unit
```

We only see this for the first map message and there does not appear to be an issue rendering the map.

KANZEZ, 2023-05-14:
Hi, any tips on how to solve this error, I met the same problem on Ubuntu22.04, ros2, humble, when trying to display a map in Rviz2

w-sribunma, 2023-11-10:
Do we have any updates on this? I am running into the same problem with rviz2

mikeferguson, 2023-11-24:
I'm not sure if this is a red herring or not - but I thought I would add some details of what I'm seeing on Rolling / 22.04 in a Parallels VM on Mac M1:

 * This error pops up for the first map message - but things continue to work as expected - UNLESS:
 * If I have a LaserScan visualization running as well, then the viewport locks up when the map is received. Visualization stops happening, can't zoom/rotate the view. It never recovers - no other error is printed. Reset button doesn't help.
 * If I disable the LaserScan visualization before the first map appears, the viewport continues to work as expected.

I can't say conclusively that this message is the cause of the viewport locking up, or if it is even related.

TheNoobInventor, 2024-01-27:
I am not sure if this will help anyone out, but I had this same error and I solved it by changing my WiFi router. I changed my router to one of those Mango mini travel routers and it fixed my issue. I used the router to create a separate network for my PC and robot to connect to.

Maaliiik, 2024-03-03:
I am running into the same problem with ros2 and rviz2 when trying to load a map through the Slam toolbox.
Do we have any updates on this?

EndlessLoops, 2025-02-28:
Same problem in ros2 jazzy.

MularX, 2025-08-14:
Got the same error as original poster however this last fix does not work for me

```
[ERROR] [1755168998.460232782] [rviz2]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : 
active samplers with a different type refer to the same texture image unit
```

https://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#cannot-start-rviz2

this didn't fix it for me as it seems this is resolving issue considering launching rviz not this grahpic function issue.
The other link seems to be for ros rolling so i can't make it work either because i work on humble.
Did anyone found a way to fix it or at least allow rviz2 to take map despite the error?
I used to not have this error what might be causing this?