ros2/rviz #464: Rviz2 marker lifetime

ZhilaiShen, 2019-09-10:
I am wondering if the "lifetime" of markers can be applied. When I try to delete the CUBE marker by using marker.lifetime = rclcpp::Duration(1), the marker is not deleted automatically from rviz. Is there something else I should try?

Martin-Idel, 2019-10-17:
The code for deletion is present, so in principle markers with expired lifetime should get deleted automatically. I had a quick look and found two fishy areas in the code, maybe you can test whether this already solves the problem? If not, I'll have to come back and investigate this further.

ZhilaiShen, 2019-10-17:
Hi Martin, sorry for the late update, but the problem is solved. I used marker.lifetime=rclcpp::Duration(100*time), where time is a static constexpr int and has value of 100000000. I thought it might relate to the time I use. I am not sure if it is a problem if I send the messages inside a Docker container and plot it using RVIZ2 under UBUNTU 16.04, but everything is working right now. Feel free to close this issue.

closed 2019-10-18 by jacobperron

jpapon, 2021-04-16:
Anyone that comes across this in the future, remember that rclcpp::Duration constructor is in nanoseconds, and if you use small integers, or worse, floats, expecting that is in seconds, you'll get weird behavior.
If you specify the value in nanoseconds, the lifetime will work correctly.
See:
https://docs.ros2.org/ardent/api/rclcpp/classrclcpp_1_1_duration.html

PR #472 (2019-10-18): Use clock from context in markers
Markers currently use their own clock to set marker lifetime. This is definitely unintended as everything within RViz should use the same clock. 

In addition, the deletion method for makers with expired lifetime looks suspicious, so I refactored it to work better. 

This might or might not be enough to fix #464, I couldn't test it right now.