ros2/rviz #469: Panel factory doesn't load plugins

SteveMacenski, 2019-10-08:
I've been trying to port a rviz panel from ROS1 to 2 and it doesn't look like any of the panels are registered as plugins, and the panel factory itself isn't making an attempt to load them either. 

I see in the [display factory](https://github.com/ros2/rviz/blob/ros2/rviz_common/src/rviz_common/display_factory.cpp#L55) it will load the plugins but the [panel factory](https://github.com/ros2/rviz/blob/ros2/rviz_common/src/rviz_common/panel_factory.cpp#L57) only loads the default builtin options without any functionality to load plugins of the panel type.

This is also in contradiction to the developer documentation which includes panel plugins as already available

Is there a timeline to getting panel plugins supported? It looks like it was just glanced over since it seems all the other plugin types are ported to support external plugins

wjwwood, 2019-10-25:
Hmm, this looks like an oversight in the porting. I didn't port this part, nor did I write the developer docks, myself, but I imagine that's what happened.

If you can see how to fix it, a pull request would be appreciated. Essentially you'd just need to remove the hard coded paths and instead load them from pluginlib as the display plugins are.

Otherwise, I think we'll have to put this in backlog with "help wanted" for now.

SteveMacenski, 2019-10-25:
Thats a good place for it. 

I may get to it, but I’ve got quite a bit on my plate until the new year

nbbrooks, 2020-01-14:
I've been following [this thread](https://answers.ros.org/question/341214/custom-panel-plugin-not-being-found-when-running-rviz2/) which points to this issue.

I'm not encountering the behavior described here - I get 2 different sets of behavior.

#1 Debian install
When I am using ros-eloquent-desktop debians and my package with a custom display and custom panel, the dialogs for adding display/panel show only the default options. From this issue, I expected to see the Display list to include my custom Display (but not the Panel dialog).

#2 Source install
After uninstalling ros-eloquent-* and installing ros2 eloquent from source, both the custom display and custom panel show up as options in their respective dialogs (see images). I didn't expect the behavior to be different than #1. 
[image]
[image]

Looking at the apt show version for my debian install it doesn't seem like much has changed.
rviz2 repo has one additional commit (looks unrelated) and pluginlib repo is at the debian version. I'm not sure if there are other libraries which may be a factor in this.

SteveMacenski, 2020-01-15:
Maybe just needs a new release these days. It's also been 3 months since I filed this.

wjwwood, 2020-01-15:
I don't think anyone has addressed the issue in the meantime. I feel like they would have referenced this issue had they done so, but maybe not.

I agree that the difference in master and what's released is unlikely to affect this. It might be a subtle bug in how the plugins are discovered? I'm sorry I haven't and still do not have time to look into it in more detail.

nbbrooks, 2020-01-15:
That is my thought as well, that the fact the source install and my rviz_example package are now in the same workspace folder is somehow circumventing the issue.

nbbrooks, 2020-01-24:
This issue had been resolved. My initial issue was incorrect documentation [here](https://github.com/ros2/rviz/blob/ros2/docs/plugin_development.md) - I made a PR to fix it.

The problem of it working with the source install but not debian was problems from having a dirty colcon/ament prefix path (some details [here](https://answers.ros.org/question/341214/custom-panel-plugin-not-being-found-when-running-rviz2/)).

My displays and panels are now both working for source/debian installs.

wjwwood, 2020-01-24:
Glad to hear it, can we close this issue?

SteveMacenski, 2020-01-24:
Sounds good. I thought I'd tried that - but I'll take @nbbrooks word for it. I won't be able to address this again for another couple weeks.

closed 2020-01-24 by SteveMacenski