ros2/rviz #476: rviz2 doesn't display a RobotModel when published with robot_state_publisher

christophebedard, 2019-10-23:
Installed from debs, version seems to be `20190923` (from the packages).

I'm trying to get a URDF model in rviz2. I have a simple launch file that creates a `robot_state_publisher` with a path to a URDF file, much like [this example](https://github.com/ROBOTIS-GIT/turtlebot3/blob/ros2/turtlebot3_bringup/launch/turtlebot3_state_publisher.launch.py#L47-L53) from the turtlebot3 repo.

I do this:

1. open rviz2
2. create a RobotModel display with the `/robot_description` topic
3. launch the `robot_state_publisher` node with the launch file mentioned above

I would expect the model to appear, but it does not, even if I create the display only after launching the node.

From looking at the source code for `robot_state_publisher` @ `dashing` (or `2.2.4`), I think this is normal, as it is only published after getting joint positions ([here](https://github.com/ros/robot_state_publisher/blob/bffb9d14c7452df0a96b92cd338c4137ec6e39b7/src/robot_state_publisher.cpp#L149), which I do not provide/publish since this is a "static" model. This was changed in the `ros2` branch to be published in the constructor [here](https://github.com/ros/robot_state_publisher/blob/cbc26e689cfce4828fe8af34e0aa4aa2d1825642/src/robot_state_publisher.cpp#L92), so I compiled this branch from source. However, it still does not work.

What *does* work, though, is the following:

1. open rviz2
2. create a RobotModel display with the `/robot_description` topic
3. launch the `robot_state_publisher` node with the launch file mentioned above
4. in another terminal: `ros2 topic echo /robot_state_publisher`

The model appears in rviz2, but the message isn't seen by `ros2 topic echo`. Interestingly, this does not work if swap steps 3 and 4.

@wjwwood mentioned that this might've gotten fixed already (haven't tried using rviz2 from source), but I can't find any related PR here or in `robot_state_publisher`.

Note: in my case, since I'm using a static model, I just switched to using a "File" description source.

wjwwood, 2019-10-23:
Also, you're using the dashing debs and fastrtps as the rmw implementation, right?

christophebedard, 2019-10-23:
> Also, you're using the dashing debs and fastrtps as the rmw implementation, right?

Yep! Sorry, should've included it.

sloretz, 2019-11-22:
I'm able to reproduce the issue using the latest `dashing` debs. I'm not able to reproduce using the latest sources or `eloquent` debs. The robot model shows up regardless which order I start `robot_state_publisher` or `rviz2`.

Using `dashing` I don't see the robot model when running the steps below. I also don't see the `robot_model` if I launch `robot_state_publisher` before `rviz2`. There's no console output from `rviz2` to indicate what the issue is.

1. `rviz2`
2. Manually set fixed frame to `base_link` (no tf data yet, must type manually)
2. Add `RobotModel` display set to `/robot_description` topic
3. `TURTLEBOT3_MODEL=waffle ros2 launch turtlebot3_bringup turtlebot3_state_publisher.launch.py`

However, I'm not able to reproduce the model appearing after running `ros2 topic echo /robot_description`. I haven't found a way to make the model appear in RViz using dashing.

sloretz, 2019-11-22:
> From looking at the source code for robot_state_publisher @ dashing (or 2.2.4), I think this is normal, as it is only published after getting joint positions (which I do not provide/publish since this is a "static" model).

I am able to get the robot model to display in Dashing if I call `ros2 topic pub /joint_states sensor_msgs/msg/JointState "{}"`, so it seems like this should be an issue on `robot_state_publisher`. Whatever was fixed there for `eloquent` should be backported to `dashing`.

clalancette, 2019-11-22:
So in dashing, robot_state_publisher doesn't publish any transforms at all until at least one `/joint_state` message is received.  This was changed in Eloquent to always publish the static transforms during the constructor.  I was somewhat hesitant to change it in Dashing, since it is a change of behavior, but it is probably innocuous.  @sloretz if you want to do a backport of that change, I can review it.

allenh1, 2019-11-22:
@sloretz / @clalancette the relevent PR is this one, I think: ros2/robot_state_publisher#24

sloretz, 2019-12-04:
https://github.com/ros/rosdistro/pull/23210

sloretz, 2019-12-04:
The fix should be in the next Dashing patch release. I'll close this for now.

closed 2019-12-04 by sloretz