ros2/rviz #477: Build fails due to an issue with libGL

cvasfi, 2019-11-06:
OS: Ubuntu 16.04
Ros2 version: Dashing
_____________

I follow the instructions on https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/ to build rviz from its source but it fails at `rviz_rendering` with the following error:

```bash
--- stderr: rviz_rendering                                                                                                                                                                 
/usr/lib/x86_64-linux-gnu/libGL.so: undefined reference to `_glapi_tls_Current'
collect2: error: ld returned 1 exit status
make[2]: *** [point_cloud_renderable_test_target] Error 1
make[1]: *** [CMakeFiles/point_cloud_renderable_test_target.dir/all] Error 2
```

Also tried building from master but got the same issue.

Martin-Idel, 2019-11-21:
Can you
a) verify that opengl is in general working (i.e. `glxgears` works correctly)
b) verify that `/usr/lib/x86_64-linux-gnu/libGL.so` actually contains the symbol?

wjwwood, 2019-11-21:
@cvasfi also, we don't support Dashing or master on 16.04, is it possible for you to try to reproduce this on 18.04?

See our supported platforms: https://www.ros.org/reps/rep-2000.html#dashing-diademata-may-2019-may-2021

wjwwood, 2019-12-05:
I'm going to close this for now since we don't support Dashing on 16.04, but @cvasfi if you have time to follow up with me or @Martin-Idel do it here and I can reopen if needed.

closed 2019-12-05 by wjwwood