ros2/rviz #478: Should map dispay default to transient local?

sloretz, 2019-11-12:
It looks like the map topic (not the updates topic) on the map display defaults to volatile durability, but is changeable with a dropdown menu. [`dummy_map_server` uses transient local durability](https://github.com/ros2/demos/blob/21f55db3c080a2d44a4d2af7c0ede0727571ee0d/dummy_robot/dummy_map_server/src/dummy_map_server.cpp#L34).

Assuming the purpose of the two topics is to have a large map come over the main topic, with smaller changes on updates, should the map display subscribe with transient local by default to make sure it gets the initial map message?

If so, it looks like this could be done by [setting `qos_profile`](https://github.com/ros2/rviz/blob/f560c940c91ff1839f6a95849575aa657f85a4da/rviz_common/include/rviz_common/ros_topic_display.hpp#L107) before calling [MFDClass::subscribe](https://github.com/ros2/rviz/blob/f560c940c91ff1839f6a95849575aa657f85a4da/rviz_default_plugins/src/rviz_default_plugins/displays/map/map_display.cpp#L206).

jacobperron, 2019-11-25:
There is a somewhat related discussion in https://github.com/ros2/rviz/issues/400.
I'm not sure what the default durability setting should be for the map topic, but in any case the user should be able to save their RViz configuration with the desired QoS setting.

jacobperron, 2019-11-25:
One argument for keeping the default as volatile is that it is compatible with both volatile and transient local publishers, whereas a transient local subscription requires a transient local publisher.

clalancette, 2019-11-26:
> One argument for keeping the default as volatile is that it is compatible with both volatile and transient local publishers, whereas a transient local subscription requires a transient local publisher.

Just FYI, this isn't strictly true.  A volatile publisher can publish just fine to a transient_local subscription.  The problem is that a transient_local subscription that comes in late won't get any of the "old" data from the transient_local publisher.

The distinction may not matter here, but I think it is important to keep in mind.

jacobperron, 2019-11-26:
>  A volatile publisher can publish just fine to a transient_local subscription.

:thinking: I tried the map display with a transient local subscription in RViz and a volatile publisher, but didn't receive any messages.

clalancette, 2019-11-27:
> > publish just fine to a transient_local subscription.
> 
> thinking I tried the map display with a transient local subscription in RViz and a volatile publisher, but didn't receive any messages.

https://github.com/clalancette/ros2_transient_local is my personal project that I use to test this stuff out.  In there there is a transient local subscriber and publisher.  If you clone and build that, you can run:

```
ros2 run transient_local transient_local_pub
```

and:

```
ros2 run transient_local transient_local_sub
```

(in that order).  You'll get the queued data in the subscriber.  In a third terminal, if you then run:

```
ros2 topic pub -1 '/latched' std_msgs/msg/String '{data: hello}'
```

You'll see the data show up in the subscriber as well.

jacobperron, 2019-11-27:
Your example works. There must be something else going on with RViz.

wjwwood, 2019-12-13:
> with smaller changes on updates

I think that's not possible, instead you just publish the whole map whenever it changes, but since it's large you don't want to send it periodically as the clients that already have it will receive it again unnecessarily. That's the reason for transient local durability in my opinion.

----

I would vote for defaulting it to transient local, and maybe instead default to the QoS of the only publisher (when that becomes possible), or if there's more than one or no publishers yet then default to transient local, or what ever the user chooses if they pick on option from the drop down.

wjwwood, 2019-12-13:
@sloretz you've been randomly selected (amongst the four of us) to "resolve" this issue, however you think most appropriate.