ros2/rviz #480: Window Empty after Screenshot

allenh1, 2019-11-20:
Rather odd, but if you do a File -> Save Image, the viewing of the rendered model seems to disappear.

[image]

The FPS is indicative of work being done, but there's an inability to see anything.

I should also point out I'm on Windows 10.

allenh1, 2019-11-20:
In the source, it appears things propagate from Ogre.

https://github.com/ros2/rviz/blob/1866827f2277ff79e7c4ff99d1612d6648ecdc6e/rviz_rendering/src/rviz_rendering/ogre_render_window_impl.cpp#L97-L101

Martin-Idel, 2019-11-21:
This seems like a duplicate of #351

allenh1, 2019-11-21:
@Martin-Idel yep. My mistake.

closed 2019-11-21 by allenh1