ros2/rviz #484: Investigate and fix timing issues when getting transforms from displays

jacobperron, 2019-11-27:
Follow up https://github.com/ros2/rviz/pull/483.

It's possible that a similar issue to https://github.com/ros2/rviz/pull/483 may occur with other displays; rendering will fail if the fixed frame is not static. This may occur if calls to the frame managers `getTransform()` or `transform()` functions fail when provided a non-zero timestamp.

Here is a list of displays that request transforms, which we should confirm render properly when in a moving frame:

- Axes
- Camera
- GridCell
- InteractiveMarkers
- Marker (marker_base.cpp)
- Odometry
- Path
- Polygon
- PointCloud (point_cloud_common.cpp)
- PointStamped
- Pose
- PoseArray
- PoseCovariance
- Range
- Wrench

olmerg, 2021-02-10:
There exist some prevision to solve this issue?

jacobperron, 2021-02-11:
> There exist some prevision to solve this issue?

I don't have bandwidth in the near-term to look at this issue. Any help is appreciated.