ros2/rviz #486: RViz crashes when mouse intersects with Image plugin panel

jacobhuesman, 2019-12-02:
Rviz crashes with the following error when my mouse intersects the portion of the screen that the Image plugin displays images

rviz2: /tmp/binarydeb/ros-eloquent-rviz-ogre-vendor-7.0.3/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreSceneManager.cpp:1384: virtual void Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool): Assertion `camVisObjIt != mCamVisibleObjectsMap.end() && "Should never fail to find a visible object bound for a camera, " "did you override SceneManager::createCamera or something?"' failed.

Otherwise the plugin appears to function normally. I checked for a similar bug with the Camera plugin, but it appears to function correctly.

I'm running ROS2 Eloquent on Ubuntu 18.04 inside a docker container.

Let me know if you need additional info.

jacobperron, 2019-12-04:
Thanks for reporting. I can reproduce the issue with the Eloquent binaries as well as building the latest source.

jacobperron, 2019-12-07:
I build with debug and get the following assertion failure within Ogre:

```
build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreSceneManager.cpp:1384: virtual void Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool): Assertion `camVisObjIt != mCamVisibleObjectsMap.end() && "Should never fail to find a visible object bound for a camera, " "did you override SceneManager::createCamera or something?"' failed.
```

<details>
<summary>Backtrace</summary>
<pre>
#0  0x00007ffff54cfe97 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff54d1801 in __GI_abort () at abort.c:79
#2  0x00007ffff54c139a in __assert_fail_base (fmt=0x7ffff56487d8 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=assertion@entry=0x7ffff5098358 "camVisObjIt != mCamVisibleObjectsMap.end() && \"Should never fail to find a visible object bound for a camera, \" \"did you override SceneManager::createCamera or something?\"", file=file@entry=0x7ffff5097f28 "/home/jacob/ws/ros2_rviz_486_ws/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreSceneManager.cpp", line=line@entry=1384, function=function@entry=0x7ffff5098b00 <Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool)::__PRETTY_FUNCTION__> "virtual void Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool)") at assert.c:92
#3  0x00007ffff54c1412 in __GI___assert_fail (assertion=0x7ffff5098358 "camVisObjIt != mCamVisibleObjectsMap.end() && \"Should never fail to find a visible object bound for a camera, \" \"did you override SceneManager::createCamera or something?\"", file=0x7ffff5097f28 "/home/jacob/ws/ros2_rviz_486_ws/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreSceneManager.cpp", line=1384, function=0x7ffff5098b00 <Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool)::__PRETTY_FUNCTION__> "virtual void Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool)") at assert.c:101
#4  0x00007ffff4f326b1 in Ogre::SceneManager::_renderScene(Ogre::Camera*, Ogre::Viewport*, bool) (this=0x555555fc1300, camera=0x555556173090, vp=0x5555561903e0, includeOverlays=false)
    at /home/jacob/ws/ros2_rviz_486_ws/build/rviz_ogre_vendor/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreSceneManager.cpp:1382
#5  0x00007ffff7a8a3fc in rviz_common::interaction::SelectionRenderer::renderToTexture(Ogre::RenderTexture*, Ogre::Viewport*) (this=0x555556003550, render_texture=0x55555602ff40, window_viewport=0x5555561903e0)
    at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_common/src/rviz_common/interaction/selection_renderer.cpp:264
#6  0x00007ffff7a89879 in rviz_common::interaction::SelectionRenderer::render(rviz_rendering::RenderWindow*, rviz_common::interaction::SelectionRectangle, rviz_common::interaction::RenderTexture, rviz_common::interaction::HandlerRange, Ogre::PixelBox&) (this=0x555556003550, window=0x5555560463e0, rectangle=..., texture=..., handlers=..., dst_box=...) at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_common/src/rviz_common/interaction/selection_renderer.cpp:111
#7  0x00007ffff7a7c8ff in rviz_common::interaction::SelectionManager::render(rviz_rendering::RenderWindow*, rviz_common::interaction::SelectionRectangle const&, rviz_common::interaction::RenderTexture const&, Ogre::PixelBox&) (this=
    0x55555600e8d0, window=0x5555560463e0, selection_rectangle=..., render_texture=..., dst_box=...) at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_common/src/rviz_common/interaction/selection_manager.cpp:309
#8  0x00007ffff7a7c74d in rviz_common::interaction::SelectionManager::renderAndUnpack(rviz_rendering::RenderWindow*, rviz_common::interaction::SelectionRectangle const&, unsigned int) (this=0x55555600e8d0, window=0x5555560463e0, selection_rectangle=..., pass=0)
    at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_common/src/rviz_common/interaction/selection_manager.cpp:298
#9  0x00007ffff7a7e380 in rviz_common::interaction::SelectionManager::pick(rviz_rendering::RenderWindow*, int, int, int, int, std::unordered_map<unsigned int, rviz_common::interaction::Picked, std::hash<unsigned int>, std::equal_to<unsigned int>, std::allocator<std::pair<unsigned int const, rviz_common::interaction::Picked> > >&) (this=0x55555600e8d0, window=0x5555560463e0, x1=277, y1=76, x2=278, y2=77, results=std::unordered_map with 0 elements)
    at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_common/src/rviz_common/interaction/selection_manager.cpp:605
#10 0x00007fffaba3ce03 in rviz_default_plugins::tools::InteractionTool::updateFocus(rviz_common::ViewportMouseEvent const&) (this=0x5555560228d0, event=...)
    at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_default_plugins/src/rviz_default_plugins/tools/interaction/interaction_tool.cpp:94
#11 0x00007fffaba3d329 in rviz_default_plugins::tools::InteractionTool::processMouseEvent(rviz_common::ViewportMouseEvent&) (this=0x5555560228d0, event=...)
    at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_default_plugins/src/rviz_default_plugins/tools/interaction/interaction_tool.cpp:159
#12 0x00007ffff7ad97fa in rviz_common::VisualizationManager::handleMouseEvent(rviz_common::ViewportMouseEvent const&) (this=0x55555589ae70, vme=...) at /home/jacob/ws/ros2_rviz_486_ws/src/rviz/rviz_common/src/rviz_common/visualization_manager.cpp:648
[... cut ...] __args#0=@0x7fffffff1a40: 0x7fffffff1d80) at /usr/include/c++/7/bits/std_function.h:316
#19 0x00007ffff3ea87e7 in std::function<void (QMouseEvent*)>::operator()(QMouseEvent*) const (this=0x555556046410, __args#0=0x7fffffff1d80) at /usr/include/c++/7/bits/std_function.h:706
</pre>
</details>

This is the call to Ogre that is failing:

https://github.com/ros2/rviz/blob/5595a69398df4d93aea711d408772ea34c53c230/rviz_common/src/rviz_common/interaction/selection_renderer.cpp#L264

Still investigating.

jacobperron, 2019-12-07:
Possible fix https://github.com/ros2/rviz/pull/490
Though there's probably a better solution.

alsora, 2020-01-29:
Any updates on this?

jacobperron, 2020-01-29:
@alsora If you can, please try #490 for a fix. I still can't explain why the crash is happening though..

closed 2020-02-03 by jacobperron

ruffsl, 2020-02-03:
Thanks @jacobperron !
Looking forward to the next sync, this killing me whenever I'd need to adjust image windows in rviz2.

PR #490 (2020-02-03): Remove uncessary call to render scene
Fixes #486.

At least the call seems unncessary. I haven't noticed any changes in selection render behavior
with or without the change. But, making the call causes an assertion to fail within Ogre when
a selection tool is moused-over an image display (see #486 for details).

The hack was originally added in

https://github.com/ros-visualization/rviz/commit/7feb02d6adf2d3a02343fd43b4b386f515591ab4

but it's not clear if the original buggy behavior still exists without it. There's been a lot
of refactoring since then. I've played around with selecting things [... cut]
commit 56211da9 (2020-02-03): Remove uncessary call to render scene (#490)

* Remove uncessary call to render scene

Fixes #486.

At least the call seems unncessary. I haven't noticed any changes in selection render behavior
with or without the change. But, making the call causes an assertion to fail within Ogre when
a selection tool is moused-over an image display (see #486 for details).

The hack was originally added in

https://github.com/ros-visualization/rviz/commit/7feb02d6adf2d3a02343fd43b4b386f515591ab4

but it's not clear if the original buggy behavior still exists without it. There's been a lot
of refactoring sin [... cut]
commit ce16e88f (2020-03-11): Remove uncessary call to render scene (#490) (#515)

* Remove uncessary call to render scene

Fixes #486.

At least the call seems unncessary. I haven't noticed any changes in selection render behavior
with or without the change. But, making the call causes an assertion to fail within Ogre when
a selection tool is moused-over an image display (see #486 for details).

The hack was originally added in

https://github.com/ros-visualization/rviz/commit/7feb02d6adf2d3a02343fd43b4b386f515591ab4

but it's not clear if the original buggy behavior still exists without it. There's been a lot
of refactor [... cut]
commit 592a0c20 (2020-03-31): Remove uncessary call to render scene (#490) (#514)

* Remove uncessary call to render scene

Fixes #486.

At least the call seems unncessary. I haven't noticed any changes in selection render behavior
with or without the change. But, making the call causes an assertion to fail within Ogre when
a selection tool is moused-over an image display (see #486 for details).

The hack was originally added in

https://github.com/ros-visualization/rviz/commit/7feb02d6adf2d3a02343fd43b4b386f515591ab4

but it's not clear if the original buggy behavior still exists without it. There's been a lot
of refactor [... cut]