ros2/rviz #492: ros2-dashing-debug windows rviz2 ERROR

cjsJack, 2019-12-11:
I am using ros2-dashing-debug in windows, using vs2017, when I command rviz2, it appear some error:
C:\Windows\System32>rviz2
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/rotate.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/rotate_cam.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/move2d.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/move_z.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/zoom.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/crosshair.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/crosshair.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/crosshair.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/rotate.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/rotate_cam.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/move2d.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/move_z.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/zoom.svg -- using default cursor instead.
[ERROR] [rviz2]: Could not load pixmap package://rviz_common/icons/crosshair.svg -- using default cursor instead.
But rviz can also succeed showing the window

wjwwood, 2020-01-02:
Did you source the `setup.bat` file? Please give more details on how to reproduce your issue.

I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2020-01-02 by wjwwood

kelebujiabing, 2020-03-28:
[image]

hi, I meet the same errs. And source the setup.bat doesn't work for me.pls help, thanks

cjsJack, 2020-03-31:
I am sorry, I don't know how to solve this problem, but  this problem
doesn't effect we use this version.