ros2/rviz #493: OSX rviz_assimp_vendor build error

claireyywang, 2019-12-11:
The file name `/usr/local/lib/libassimp.dylib.5` is likely a typo, I think what it meant to be is `/usr/local/lib/libassimp.5.dylib`, which is the file I found in the directory.

A temporary fix is to create symbolic link `ln -s /usr/local/lib/libassimp.5.dylib /usr/local/lib/libassimp.dylib.5`

The following is the error message I got.

```bash
CMake Error at /usr/local/lib/cmake/assimp-5.0/assimpTargets.cmake:85 (message):
  The imported target "assimp::assimp" references the file

     "/usr/local/lib/libassimp.dylib.5"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/local/lib/cmake/assimp-5.0/assimpTargets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/local/lib/cmake/assimp-5.0/assimp-config.cmake:1 (include)
  CMakeLists.txt:55 (find_package)

---
Failed   <<< rviz_assimp_vendor	[ Exited with code 1 ]
```

wjwwood, 2019-12-11:
@nuclearsandwich I think we are intentionally holding back assimp to 4 (And not 5) on the buildfarm, but I can't find any build cop or other issue about it, is there one?

nuclearsandwich, 2019-12-12:
> I think we are intentionally holding back assimp to 4 (And not 5) on the buildfarm, but I can't find any build cop or other issue about it, is there one?

We haven't actually been intentionally holding it back until Monday as before then we were just behind on upgrades. @gerkey submitted a patch upstream which has been accepted and not yet released, I'm working on getting homebrew to take the the patch in the meantime. Will link a PR when it is ready.

wjwwood, 2019-12-12:
The contribution by @gerkey that was mentioned: https://github.com/assimp/assimp/pull/2765

Thanks for the update @nuclearsandwich.

nuclearsandwich, 2019-12-12:
https://github.com/Homebrew/homebrew-core/pull/47792 is open with my attempt to add the patch to Homebrew. If that goes through we'll be able to update our mac hosts.

claireyywang, 2019-12-13:
Thanks for the follow up and update! I'll keep track of this issue.

nuclearsandwich, 2019-12-16:
The patch above has been merged but until the binary bottle is rebuilt I think you'll need to use `brew reinstall assimp --build-from-source` to get the patched behavior.

yathartha3, 2019-12-25:
@nuclearsandwich ```brew reinstall assimp --build-from-source``` worked for me :)

claireyywang, 2019-12-30:
> The patch above has been merged but until the binary bottle is rebuilt I think you'll need to use `brew reinstall assimp --build-from-source` to get the patched behavior.

Thanks @nuclearsandwich ! Do you know when the binary bottle will be rebuilt? Is the wait long enough to justify adding this issue to the `troubleshoot` section of installation documentation?

claireyywang, 2020-03-06:
error went away with newly built bottle

closed 2020-03-06 by claireyywang