ros2/rviz #494: rviz2 crashes when trying to visualize RobotModel

pbeeson, 2019-12-19:
Using ros2 eloquent from debs, I have tried both listening to a URDF published by robot_State_publisher on the /robot_description topic, and I've tried loaded the URDF directly from file for the Rviz RobotModel plugin.  In both cases, I get this error:
`
Parsing robot urdf xml string.
[ERROR] [rviz2]: ItemIdentityException: Cannot find a group named rviz_common in ResourceGroupManager::isResourceGroupInitialised at /tmp/binarydeb/ros-eloquent-rviz-ogre-vendor-7.0.3/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 1770)
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  ItemIdentityException: Cannot find a group named rviz_common in ResourceGroupManager::isResourceGroupInitialised at /tmp/binarydeb/ros-eloquent-rviz-ogre-vendor-7.0.3/obj-x86_64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreResourceGroupManager.cpp (line 1770)
Aborted (core dumped)
`

I am using a urdf that was created by running xacro.py on the old pr2.urdf.xacro and dumping the output to a static, large urdf.  This urdf works fine with Rviz1, the ros2 urdf_parser handles it, etc.  Just RViz2 bombs out.  I haven't tried any other URDFs, but I'm about to, and will follow up.

pbeeson, 2019-12-19:
`/opt/ros/eloquent/share/rviz_default_plugins/test_meshes/test.urdf` works.

pbeeson, 2019-12-19:
So for some reason rviz2 is bombing on a urdf that is parsable and visible in rviz1.

Here's the URDF.  Any thoughts on why this particular urdf is bombing rviz would be appreciated.  Again, I can parse this and use the collison models with my FCL-based collision library.  No problems.  It's just Rviz that doesn't seem to like it.

```
<?xml version="1.0" encoding="utf-8"?>
<robot name="pr2" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <material name="Blue">
    <color rgba="0.0 0.0 0.8 1.0"/>
  </material>
  <material name="Green">
    <color rgba="0.0 0.8 0.0 1.0"/>
  </material>
  <material name="Grey">
    <color rgba="0.7 0.7 0.7 1.0"/>
  </material>
  <material name="Grey2">
    <color rgba="0.9 0.9 0.9 1.0"/>
  </material>
  <material name="Red">
    <color rgba="0.8 0.0 0.0 1.0"/>
  </material>
  <material name="White">
    <color rgba="1.0 1.0 1.0 1.0"/>
  </material>
  <material name="Black">
    <color rgba="0.1 0.1 0.1 1.0"/>
  </material>
  <material name="LightGrey">
    <color rgba="0.6 0.6 0.6 1.0"/>
  </material>
  <material name="Caster">
    <texture filename="package://pr2_description/materials/textures/pr2_caster_texture.png"/>
  </material>
  <material name="Wheel_l">
    <texture filename="package://pr2_description/materials/textures/pr2_wheel_left.png"/>
  </material>
  <material name="Wheel_r">
    <texture filename="package://pr2_description/materials/textures/pr2_wheel_right.png"/>
  </material>
  <material name="RollLinks">
    <texture filename="package://pr2_description/materials/textures/pr2_wheel_left.png"/>
  </material>
  <!-- Now we can start using the macros included above to define the actual PR2 -->
  <link name="base_link">
    <inertial>
      <mass value="116.0"/>
      <origin xyz="-0.061 0.0 0.1465"/>
      <inertia ixx="5.652232699207" ixy="-0.009719934438" ixz="1.293988226423" iyy="5.669473158652" iyz="-0.007379583694" izz="3.683196351726"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://pr2_description/meshes/base_v0/base.stl"/>
      </geometry>
      <material name="White"/>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
        <mesh filename="package://pr2_description/meshes/base_v0/base_L.stl"/>
      </geometry>
    </collision>
  </link>
  <!-- base_footprint is a fictitious link(frame) that is on the ground right below base_link origin,
         navigation stack dedpends on this frame -->
  <link name="base_footprint">
    <inertial>
      <mass value="1.0"/>
      <origin xyz="0 0 0"/>
[... 2234 lines cut ...]
    <origin rpy="-1.57079632679 0.0 -1.57079632679" xyz="0 0 0"/>
    <parent link="l_forearm_cam_frame"/>
    <child link="l_forearm_cam_optical_frame"/>
  </joint>
  <link name="l_forearm_cam_optical_frame"/>
  <joint name="r_forearm_cam_frame_joint" type="fixed">
    <origin rpy="1.57079632679 -0.562868683768 0" xyz=".135 0 .044"/>
    <parent link="r_forearm_roll_link"/>
    <child link="r_forearm_cam_frame"/>
  </joint>
  <link name="r_forearm_cam_frame">
    <inertial>
      <mass value="0.01"/>
      <origin xyz="0 0 0"/>
      <inertia ixx="0.001" ixy="0.0" ixz="0.0" iyy="0.001" iyz="0.0" izz="0.001"/>
    </inertial>
  </link>
  <joint name="r_forearm_cam_optical_frame_joint" type="fixed">
    <origin rpy="-1.57079632679 0.0 -1.57079632679" xyz="0 0 0"/>
    <parent link="r_forearm_cam_frame"/>
    <child link="r_forearm_cam_optical_frame"/>
  </joint>
  <link name="r_forearm_cam_optical_frame"/>
</robot>
```

pbeeson, 2019-12-19:
Of course, you have to then use the texture on a link.  Also, the package:// isn't the culprit as I tried absolute paths and that doesn't work either.

Martin-Idel, 2019-12-19:
No, the error message tells us that the `Ogre::ResourceManager` is upset because it doesn't find the resource group "rviz_common" (which was never initialized). There were some changes to how this is done internally at Ogre and also there is still no definitive way for RViz to handle it in ros2 (ros 1 rviz always used the package name, but there are multiple packages here which doesn't work).

This https://github.com/ros2/rviz/pull/495 should fix it.

pbeeson, 2019-12-23:
I have validated that PR #495 does indeed fix this problem.

PR #495 (2020-01-15): Fix wrong resource group for robot links
Fix #494 

We only initialize the resource groups "rviz_rendering" and the default group, thus "rviz_common" is never initialized and throws the error. Since all meshes will be put into the "rviz_rendering" resource group if they are loaded by the mesh loader, the robot meshes should be too.

It would be great to have a test, but it needs materials and I don't have any, so I'm going to leave it as is.
closed 2020-01-15 by wjwwood