ros2/rviz #497: ROS2 build failure on Mac OS Catalina @ament_cmake_pep257

TMmichi, 2020-01-06:
## Bug report

**Required Info:**
- Operating System:
  - Mac OS Catalina 10.15.2
(Although I understand that the required operating system is Sierra, I tried on Catalina.)
- Installation type:
  - from source
- Version or commit hash:
  - ROS 2 Eloquent Elusor from ros2.repos
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Followed all the instructions from Install Prerequisites except for the optional parts.

Got ros2 source files from repos
building from colcon failed at ament_cmake_pep257
```
cd ~/ros2_eloquent/
colcon build --symlink-install
```

#### Expected behavior
Building completion from colcon.

#### Actual behavior
Failed at ament_cmake_pep257

#### Additional information

```
colcon build --symlink-install
[3.634s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/Users/<>/Workspace/ros2_eloquent' in the environment variable AMENT_PREFIX_PATH doesn't contain any 'local_setup.*' files.
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr
Starting >>> foonathan_memory_vendor
Starting >>> poco_vendor
Starting >>> cyclonedds
Starting >>> ament_cppcheck
Starting >>> tinydir_vendor
Starting >>> gtest_vendor
Starting >>> osrf_pycommon
Starting >>> tinyxml2_vendor
Starting >>> orocos_kdl
Finished <<< ament_package [2.70s]
Starting >>> ament_cmake_core
Finished <<< ament_lint [2.80s]
Starting >>> ament_flake8
Finished <<< ament_cppcheck [2.85s]
Starting >>> tinyxml_vendor
Finished <<< poco_vendor [3.03s]
Finished <<< gtest_vendor [3.05s]
Starting >>> gmock_vendor
Starting >>> urdfdom_headers
Finished <<< fastcdr [3.52s]
Finished <<< tinydir_vendor [3.51s]
Finished <<< tinyxml2_vendor [3.52s]
Finished <<< osrf_pycommon [3.59s]
Starting >>> osrf_testing_tools_cpp
Starting >>> ament_pep8
Starting >>> sqlite3_vendor
Finished <<< foonathan_memory_vendor [3.92s]
Starting >>> fastrtps
Finished <<< ament_flake8 [2.36s]
Finished <<< ament_cmake_core [2.48s]
[... 531 lines cut ...]
[26.000s] WARNING:asyncio:Loop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 21496 is closed
[26.078s] WARNING:asyncio:Loop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 21502 is closed
[26.103s] WARNING:asyncio:Loop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 21506 is closed
Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x1106599d0>
Traceback (most recent call last):
  File "/Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/base_subprocess.py", line 126, in __del__
    self.close()
  File "/Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/base_subprocess.py", line 104, in close
    proto.pipe.close()
  File "/Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/unix_events.py", line 536, in close
    self._close(None)
  File "/Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/unix_events.py", line 560, in _close
    self._loop.call_soon(self._call_connection_lost, exc)
  File "/Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/base_events.py", line 711, in call_soon
[26.168s] WARNING:asyncio:Loop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 21457 is closed
[26.168s] WARNING:asyncio:Loop <_UnixSelectorEventLoop running=False closed=True debug=False> that handles pid 20739 is closed
    self._check_closed()
  File "/Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/base_events.py", line 504, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
[26.169s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending name='Task-1355' coro=<_communicate_and_close_fds() done, defined at /Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/site-packages/colcon_core/subprocess.py:262> wait_for=<_GatheringFuture pending cb=[<TaskWakeupMethWrapper object at 0x111df8880>()]>>
[26.169s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending name='Task-1356' coro=<Process.wait() done, defined at /Users/<>/anaconda3/envs/ros2_ws/lib/python3.8/asyncio/subprocess.py:133> wait_for=<Future pending cb=[<TaskWakeupMethWrapper object at 0x111e5ab80>()]>>
```

TMmichi, 2020-01-06:
@dirk-thomas Thank you for letting me know that there is a python version mismatch. After re-creating my conda env with python 3.6, I tried building it again, and received these issues.

It indeed encountered problem with rviz_assimp_vendor, and the following is the error message.

```
 colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr
Starting >>> foonathan_memory_vendor
Starting >>> poco_vendor
Starting >>> cyclonedds
Starting >>> ament_cppcheck
Starting >>> tinydir_vendor
Starting >>> gtest_vendor
Starting >>> osrf_pycommon
Starting >>> tinyxml2_vendor
Starting >>> orocos_kdl
Finished <<< ament_package [3.25s]
Finished <<< ament_lint [3.29s]
Starting >>> ament_cmake_core
Starting >>> ament_flake8
Finished <<< ament_cppcheck [3.49s]
Starting >>> tinyxml_vendor
Finished <<< fastcdr [3.60s]
Finished <<< foonathan_memory_vendor [3.71s]
Finished <<< poco_vendor [3.78s]
Finished <<< tinydir_vendor [3.77s]
Finished <<< gtest_vendor [3.80s]
Starting >>> fastrtps
Starting >>> gmock_vendor
Starting >>> urdfdom_headers
Starting >>> osrf_testing_tools_cpp
Starting >>> ament_pep8
Finished <<< cyclonedds [4.14s]
Finished <<< osrf_pycommon [4.40s]
Starting >>> sqlite3_vendor
Finished <<< tinyxml2_vendor [4.47s]
Finished <<< orocos_kdl [4.55s]
Finished <<< tinyxml_vendor [2.50s]
Finished <<< ament_flake8 [2.91s]
Finished <<< gmock_vendor [2.17s]
Finished <<< ament_cmake_core [3.07s]
Finished <<< urdfdom_headers [2.28s]
Finished <<< sqlite3_vendor [1.90s]
Starting >>> ament_cmake_libraries
Starting >>> ament_cmake_export_libraries
Starting >>> ament_cmake_include_directories
Starting >>> ament_cmake_python
Starting >>> ament_cmake_export_definitions
Starting >>> ament_cmake_export_include_directories
Starting >>> ament_cmake_export_link_flags
Starting >>> ament_cmake_version
Starting >>> ament_pep257
Finished <<< osrf_testing_tools_cpp [2.81s]
Finished <<< ament_pep8 [2.89s]
Starting >>> test_interface_files
Starting >>> ament_mypy
Finished <<< ament_cmake_libraries [2.12s]
Starting >>> ament_cmake_export_dependencies
Finished <<< ament_cmake_export_include_directories [2.05s]
Finished <<< ament_cmake_include_directories [2.18s]
Finished <<< ament_cmake_version [1.98s]
Finished <<< ament_cmake_python [2.18s]
Finished <<< ament_cmake_export_link_flags [2.12s]
Finished <<< ament_pep257 [2.11s]
Finished <<< ament_cmake_export_definitions [2.31s]
Finished <<< test_interface_files [1.98s]
Finished <<< ament_cmake_export_libraries [2.56s]
Finished <<< fastrtps [5.04s]
[... 219 lines cut ...]
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreASTCCodec.cpp.o] Error 1
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimable.cpp.o] Error 1
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationTrack.cpp.o] Error 1
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationState.cpp.o] Error 1
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimation.cpp.o] Error 1
1 error generated.
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAxisAlignedBox.cpp.o] Error 1
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAutoParamDataSource.cpp.o] Error 1
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreBillboard.cpp.o] Error 1
1 error generated.
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreBillboardChain.cpp.o] Error 1
make[4]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-build] Error 2
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
---
Aborted  <<< rviz_ogre_vendor

Summary: 64 packages finished [1min 12s]
  1 package failed: rviz_assimp_vendor
  8 packages aborted: ament_uncrustify launch_xml launch_yaml rviz_ogre_vendor shared_queues_vendor spdlog_vendor urdfdom yaml_cpp_vendor
  3 packages had stderr output: rviz_assimp_vendor rviz_ogre_vendor shared_queues_vendor
  225 packages not processed
```

klintan, 2020-01-13:
This is a known issue in version 5.0 of assimp lib, see here https://github.com/assimp/assimp/issues/2760

Quick fix as suggested:

```
ln -s /usr/local/lib/libassimp.5.dylib /usr/local/lib/libassimp.dylib.5
```

ivanpauno, 2020-01-23:
There is currently not support for Mac OS Catalina, so I will proceed to close.

closed 2020-01-23 by ivanpauno