ros2/rviz #505: Ros2 Eloquent on Mac Mojave building Error - package 'rviz_ogre_vendor' failed

meiTob, 2020-02-03:
**Required Info:**

- Operating System:
  - MacOs 10.4.6 Mojave

- Installation type:
  - from source (as binary wasn´t working either. It had some load entry point errors and someone told me to build from source

- Version or commit hash:
  -  Eloquent

- DDS implementation:
  - don´t realy know

- Client library (if applicable):
  - rviz_ogre_vendor

#### Steps to reproduce issue

- I completly clean installed Mojave and before doing anything else followed this guide:
https://index.ros.org/doc/ros2/Installation/Eloquent/OSX-Development-Setup/

The step: 
**cd ~/ros2_eloquent/
colcon build --symlink-install**
gives me Errors
#### Expected behavior
Building success

#### Actual behavior

Im getting following Error Messages:

```
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr
Starting >>> foonathan_memory_vendor
Finished <<< ament_package [2.30s]                             
Starting >>> ament_cmake_core                               
Finished <<< foonathan_memory_vendor [2.60s]                
--- stderr: fastcdr                                         
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: for: /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib/libfastcdr.1.0.10.dylib (for architecture x86_64) option "-add_rpath /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib" would duplicate path, file already has LC_RPATH for: /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib
---
Finished <<< fastcdr [2.73s]
Starting >>> fastrtps                                       
Starting >>> poco_vendor
Finished <<< ament_lint [2.90s]                             
Starting >>> ament_flake8
Finished <<< ament_cmake_core [1.87s]                                     
Starting >>> ament_cmake_libraries                                   
Finished <<< poco_vendor [1.86s]                                  
Starting >>> ament_cmake_export_libraries                         
Finished <<< ament_cmake_libraries [0.88s]                        
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_export_libraries [0.72s]                 
Starting >>> ament_cmake_python                                        
Finished <<< ament_flake8 [3.05s]                                      
Starting >>> ament_cmake_export_dependencies
Finished <<< fastrtps [3.27s]
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_include_directories [1.40s]
Starting >>> ament_cmake_export_include_directories                          
Finished <<< ament_cmake_export_dependencies [0.72s]                         
Finished <<< ament_cmake_python [1.35s]
Finished <<< ament_cmake_export_definitions [0.81s]
Starting >>> ament_cmake_test
Starting >>> ament_cmake_export_link_flags
Starting >>> ament_cmake_version         
Finished <<< ament_cmake_export_include_directories [1.36s]
[... 145 lines cut ...]
#   include "Availability.h"
            ^~~~~~~~~~~~~~~~
1 error generated.
make[5]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreASTCCodec.cpp.o] Error 1
make[4]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-build] Error 2
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
Aborted  <<< resource_retriever                          
Aborted  <<< urdf                                        
Aborted  <<< rcutils                                                 

Summary: 83 packages finished [1min 15s]
  1 package failed: rviz_ogre_vendor
  3 packages aborted: rcutils resource_retriever urdf
  3 packages had stderr output: fastcdr orocos_kdl rviz_ogre_vendor
  211 packages not processed

```

Does anyone know what causes this error, and what I need to do to get rid off it. Or should i maybe install another distribution? I know there are already similar issues already, but they all seem to work now, while Im still getting the error. I am kind of a Neewbie, if this Post here don´t belong here feel free to tell me, and maybe tell me where else I could get help.

meiTob, 2020-02-03:
> I had the same problem. However, it was solved by the following command.
> 
> `export SDKROOT="$(xcrun --sdk macosx --show-sdk-path)"`
> 
> I hope your problem will be solved.

I mean it did something, now I am able to build 100 Packages instead of 83. The finnish message is:  
```
Failed   <<< pluginlib	[ Exited with code 2 ]
Aborted  <<< rosidl_typesupport_fastrtps_cpp             
Aborted  <<< rmw_fastrtps_shared_cpp                     
Aborted  <<< rviz_ogre_vendor                                                
                                           
Summary: 96 packages finished [21min 48s]
  1 package failed: pluginlib
  3 packages aborted: rmw_fastrtps_shared_cpp rosidl_typesupport_fastrtps_cpp rviz_ogre_vendor
  3 packages had stderr output: fastcdr orocos_kdl pluginlib
  198 packages not processed
```

The full Protokoll is:
 Im now going to google, to look if I find a solution. If anyone could help, I would be realy gratefull!
```
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr
Starting >>> foonathan_memory_vendor
Finished <<< foonathan_memory_vendor [2.84s]                   
Starting >>> poco_vendor                                       
Finished <<< ament_package [3.03s]                             
Starting >>> ament_cmake_core
Finished <<< ament_lint [3.31s]                             
Starting >>> ament_flake8                                             
--- stderr: fastcdr                                                   
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: for: /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib/libfastcdr.1.0.10.dylib (for architecture x86_64) option "-add_rpath /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib" would duplicate path, file already has LC_RPATH for: /Users/tobiasmeinert/ros2_eloquent/install/fastcdr/lib
---
Finished <<< fastcdr [3.93s]
Starting >>> fastrtps                                              
Finished <<< poco_vendor [1.78s]                                     
Finished <<< ament_cmake_core [1.66s]                                     
Starting >>> ament_cmake_libraries
Starting >>> ament_cmake_export_libraries                     
Finished <<< ament_cmake_libraries [0.65s]                    
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_export_libraries [0.77s]             
Finished <<< ament_flake8 [2.36s]                             
Starting >>> ament_cmake_python                                 
Starting >>> ament_cmake_export_dependencies
Finished <<< ament_cmake_include_directories [0.95s]                 
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_export_dependencies [0.81s]               
Finished <<< ament_cmake_python [0.94s]                            
Starting >>> ament_cmake_test                                      
Starting >>> ament_cmake_export_include_directories
Finished <<< fastrtps [3.26s]                                           
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_export_definitions [1.00s]
Finished <<< ament_cmake_export_include_directories [0.77s]                
Starting >>> ament_cmake_version
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_test [0.98s]                                      
Starting >>> ament_cmake_target_dependencies
Finished <<< ament_cmake_export_link_flags [1.08s]
Starting >>> ament_pep257
Finished <<< ament_cmake_version [1.00s]                                      
Finished <<< ament_cmake_export_interfaces [1.03s]
Finished <<< ament_cmake_target_dependencies [0.89s]
Starting >>> ament_cmake
Starting >>> cyclonedds
Starting >>> ament_cppcheck                                    
[... 300 lines cut ...]
In file included from /Users/tobiasmeinert/ros2_eloquent/src/ros/pluginlib/pluginlib/test/utest.cpp:33:
In file included from /Users/tobiasmeinert/ros2_eloquent/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader.hpp:370:
/Users/tobiasmeinert/ros2_eloquent/src/ros/pluginlib/pluginlib/include/pluginlib/./class_loader_imp.hpp:628:47: error: 'operator/' is unavailable: introduced in macOS 10.15
      std::string package_file_path = (parent / "package.xml").string();
                                              ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/filesystem:1175:41: note: 'operator/' has been explicitly marked unavailable here
  friend _LIBCPP_INLINE_VISIBILITY path operator/(const path& __lhs,
                                        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/pluginlib_utest.dir/test/utest.cpp.o] Error 1
make[1]: *** [CMakeFiles/pluginlib_utest.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
---
Failed   <<< pluginlib	[ Exited with code 2 ]
Aborted  <<< rosidl_typesupport_fastrtps_cpp             
Aborted  <<< rmw_fastrtps_shared_cpp                     
Aborted  <<< rviz_ogre_vendor                                                
                                           
Summary: 96 packages finished [21min 48s]
  1 package failed: pluginlib
  3 packages aborted: rmw_fastrtps_shared_cpp rosidl_typesupport_fastrtps_cpp rviz_ogre_vendor
  3 packages had stderr output: fastcdr 
```

meiTob, 2020-02-03:
I think I found someone with a really similiar Problem: [(https://github.com/ros-perception/image_common/issues/139)]

> Hey, I think I may have figured it out. The issue seems to be that XCode has turned on the C++17 features but not the C++17 standard library in versions less than macOS 10.15. Therefore to be able to compile in Mojave and older, you will need to add the -mmacosx-version-min=10.15 flag in your CMakeLists.txt file as such:
> 
> if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
>   add_compile_options(-Wall -Wextra -Wpedantic -mmacosx-version-min=10.15)
> endif()

But I don´t know which CMakelist.txt colcon build is using, so I am not able to fix the Problem the same way he did.

jacobperron, 2020-02-04:
I've triggered a macOS CI build with Eloquent development branches (as per the install from source instructions) on one of our Mojave machines: [image]

meiTob, 2020-02-05:
So my Mac completly crashed and I wasn´t able to boot into my Mac. So i erased my Hard Drive, installed Mac OS again and then updatet to Mac Catalina and began the building process again. I am still getting some errors, but it builded completly and i am able to run the examples. So we can close this issue. But It may be relevant for someone that i was only able to install on Catalina which isn´t even listed in the Ros2 Wiki.

closed 2020-02-05 by meiTob