ros2/rviz #511: RViz2 not showing sensor_msgs/Image

mmatthe, 2020-02-26:
Using latest ROS2 release on Ubuntu 18.04 

```
$ apt-get update
$ apt-get install --only-upgrade ros-eloquent-\*
```

RViz2 Image display does not show an image (only showing "No Image")

[image]

[image]

There sometimes occurs a log message in the RViz stdout/stderr (however, these message occur sporadically and do not appear for every transmitted frame). E.g. the filter messages below are all that came out of the RViz session. No more message are shown, even though the cam2image proc still ran for frame numbers larger than 7024

```
mmatthe@NB-SLF-0015-MMA:~ $ ros2 run rviz2 rviz2 
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz]: Message Filter dropping message: frame '6840' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6841' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6842' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6843' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6844' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6845' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6870' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6871' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6872' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6873' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6874' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6900' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6901' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6902' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6903' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6904' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6930' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6931' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6932' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6933' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6934' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6960' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6961' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6962' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6963' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6964' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6990' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6991' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6992' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6993' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '6994' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '7020' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '7021' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '7022' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '7023' at time 0,000 for reason 'Unknown'
[INFO] [rviz]: Message Filter dropping message: frame '7024' at time 0,000 for reason 'Unknown'
```

The Image transmission itself is verified with

```
$ ros2 run image_tools showimage
$ ros2 run image_tools cam2image
```

The showimage node shows the image from the webcam attached to my PC.
[image]

gonzodepedro, 2020-03-17:
Created https://github.com/ros2/demos/issues/432

gonzodepedro, 2020-03-17:
To use the camera plugin in rviz you need a couple of things.

1. sensor_msgs::Image messages published to a topic with proper frame_id information.
2. sensor_msgs::CameraInfo messages published to a topic with proper frame_id information.
3. A valid transformation tree.

The first item will get fixed by ros2/demos#432. The other items need to be provided by other means. 'ros2 run image_tools showimage ' won't work with rviz by itself.

closed 2020-03-17 by gonzodepedro

hahasofia, 2020-04-21:
Hi mmatthe,
Have you solved this issue? I met the same error log "dropping message for reason 'Unknown'. :(
@mmatthe

hahasofia, 2020-04-21:
@gonzodepedro 
Sorry, I'm a new user, could you explain more about "A valid transformation tree"? Could you give a sample? Thank you very much.

Martin-Idel, 2020-04-26:
@hahasofia The issue will be fixed with the next release but cannot really be fixed now. 

The problem is the following:
To show something in RViz, RViz needs a valid transformation between the fixed frame (e.g. "map", configurable in the GUI) and the message - otherwise it's unclear where to render the message content. That information is contained in a tree (e.g.: there are three frames: world, robot body, robot head and we have a transformation between word <-> body and between body <-> head, that's a tree structure).  Normally, you have some publishers publishing this information. 

In the case of the cam2image publisher, it used to change the frame id on every message, so in order to fix this, you'd have to publish a new transformation between each new frame at the time of the message. This lets the transformation tree explode and I'm not sure it will be very performant.

Now in principle, the transformation information would not be necessary for the image display as the image display just displays an image and doesn't position it anywhere in the world. However, you usually wouldn't use RViz to just display an image message and if you have several displays, treating transformation equally seems more consistent and might reduce confusion.

hahasofia, 2020-04-29:
@Martin-Idel 
Thanks for your reply.

LMerCy, 2021-10-19:
> @hahasofia The issue will be fixed with the next release but cannot really be fixed now.
>
> The problem is the following: To show something in RViz, RViz needs a valid transformation between the fixed [...]

The frame_id is a string "camera_id" in foxy version, but it still cannot display the image, dou you have any solution？

frankkusters, 2021-12-15:
> @gonzodepedro Sorry, I'm a new user, could you explain more about "A valid transformation tree"? Could you give a sample? Thank you very much.

I've struggled with this as well. @Martin-Idel's comment provided the clues:

You need a transformation from the fixed frame (which is called `map` by default in RViz) to the message frame. To find the message frame id from `/image_raw`:

    ros2 topic echo /image_raw --no-arr

You'll see something like this:

    header:
      stamp:
        sec: 1639564328
        nanosec: 59671685
      frame_id: camera    <---- frame_id here
    height: 480
    width: 640
    encoding: rgb8
    is_bigendian: 0
    step: 1920
    data: '<sequence type: uint8, length: 921600>'

So to generate a transformation from `map` (default RViz fixed frame) to `camera` (check frame_id above, the text 'camera' will be different depending on your camera driver and settings), you can run:

    ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 map camera

Then RViz will start showing the camera.