ros2/rviz #512: rviz2 - camera plugin error with pitch != width*BBP

MCP-TTC, 2020-03-02:
Hi,

I noticed that rviz2 cannot display image messages if pitch != width * bytesPerPixel.

Given these basic information about the system:
- Ubuntu 18.04
- ROS eloquent
- default DDS

I tried the following configurations:

1. Image msg 1928x1208x4 (BGRA) -> pitch = 7936 (where 1928*4 = 7712)
rviz2 reports: **[ERROR] [rviz2]: Error loading image: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-eloquent-rviz-ogre-vendor-7.0.4/obj-aarch64-linux-gnu/ogre-v1.12.1-prefix/src/ogre-v1.12.1/OgreMain/src/OgreImage.cpp (line 227)**
But it rqt_image_view can display the image without any error.

2. Image msg 1920x1208x4 (BGRA) -> pitch = 7680
no problems with both rviz2 and rqt_image_view

I tried also different resolutions and the problem seems to be related to the pitch != width*BBP.

wjwwood, 2020-03-12:
Would you expect this to work? Seems like reasonable error checking to me. Do you consider this a bug or would you like some other behavior?

MCP-TTC, 2020-03-13:
I think this is a bug since rqt_image_view can display the image which is still valid with a pitch that is different from width * bpp due to memory alignment or whatever, but rviz2 cannot display it and gives an exception.

I didn't take a look yet at the source code of the two packages (rqt_image_view and rviz2) to check how the image is handled.

claireyywang, 2020-03-19:
@wjwwood friendly ping

Martin-Idel, 2020-03-23:
I guess it makes sense to allow images with stride. 

---

Some considerations though:

It might not be completely trivial to implement this as Ogre seems to not have an API to load an image with padding (which seems reasonable as size is an issue for gaming and visualization and the image will have to be loaded to the GPU memory for processing anyways so I guess most image will not use padding). 

So either we'd have to transform the image beforehand or if that tanks performance try to introduce the API to Ogre...

It's also not a good option to use OpenCV and/or QImage for displaying the image here, because other than large changes, this would mean the camera display (which has to use Ogre) behaves differently than the image display, which is odd from a UX perspective

gbalke, 2021-09-22:
I'm a bit confused here so I'd like to clarify some terminology:

- Pitch: describes the bytes per row
- Stride: Bytes per pixel (BPP)
- Step: ["Full row length in bytes"](https://github.com/ros2/common_interfaces/blob/master/sensor_msgs/msg/Image.msg?plain=1#L25)

If pitch == step in this context, I don't see why an improperly defined image should still render in Ogre? The Image data should provide at least as much data as is stated by the data structure that contains it. If there is more, it should be ignored. Ideally, this should also throw an error or warn as to avoid the user accidentally viewing incomplete data.

Looking at the actual code, it looks like pitch/step isn't even used and, instead, the pixel format is used to determine the size of the image (by implicitly defining stride). Since pitch can be derived by two necessary fields, you could consider the image message over-constrained. It may be reasonable to ignore the pitch given the format defines more than just the stride (also color order). However, this is a larger discussion.

https://github.com/ros2/rviz/blob/c7ade575c1de06ad07d44da7fcb34e7e0f885fff/rviz_default_plugins/src/rviz_default_plugins/displays/image/ros_image_texture.cpp#L466-L476

@MCP-TTC to clarify, is your pixel format correct for your image? Is there something else I'm missing?