ros2/rviz #513: Many signalFailure - Drop message warnings on startup

SteveMacenski, 2020-03-02:
Hi,

In the same theme of https://github.com/ros-planning/navigation2/issues/949, RViz will produce alot of these dropped message warnings once everything is activated and just getting started. After startup, they go away, but it causes confusion for new users and generally seems like something that should be fixed.

From Michael: 

> In the obstacle layer there is a MessageFilter created when the plugin is initialized during onInitialize. All of the costmap_2d_ros plugins are initialized during costmap_2d_ros's on_configure method. When on_activate is invoked, the code is waiting for the transform to be available (base_link to map). However, the transform is not available yet because the user hasn't entered the initial_pose. So, the message filter's queue fills up and the error messages start when the message filter starts dropping messages. Once the user enters the initial pose, the on_activate method continues and the queue begins to drain.

Translating that to Rviz: the message filter is created, which automatically subscribes, and fills the queue before useful. I fix this in Navigation2 by unsubscribing once the topic is created and then subscribing when it actually starts being used. See https://github.com/ros-planning/navigation2/pull/1574

SteveMacenski, 2020-03-02:
Related / more context than https://github.com/ros2/rviz/issues/443

Ryanf55, 2023-01-25:
Can we at least throttle it? I'm getting it at 10Hz, makes it hard to see any other log content. 
```
[rviz2-5] [INFO] [1674612885.459650696] [rviz]: Message Filter dropping message: frame 'base_scan' at time 140.564 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612885.523260716] [rviz]: Message Filter dropping message: frame 'odom' at time 136.749 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612885.523386495] [rviz]: Message Filter dropping message: frame 'odom' at time 140.667 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612885.650110920] [rviz]: Message Filter dropping message: frame 'base_scan' at time 140.763 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612885.747162271] [rviz]: Message Filter dropping message: frame 'odom' at time 140.871 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612885.875493196] [rviz]: Message Filter dropping message: frame 'base_scan' at time 140.963 for reason 'discarding message because the queue is full'
[component_container_isolated-6] [INFO] [1674612885.932109373] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-5] [INFO] [1674612885.939653241] [rviz]: Message Filter dropping message: frame 'odom' at time 141.041 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612886.070139848] [rviz]: Message Filter dropping message: frame 'base_scan' at time 141.163 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612886.131094368] [rviz]: Message Filter dropping message: frame 'odom' at time 137.334 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612886.131269299] [rviz]: Message Filter dropping message: frame 'odom' at time 141.245 for reason 'discarding message because the queue is full'
[rviz2-5] [INFO] [1674612886.259987441] [rviz]: Message Filter dropping message: frame 'base_scan' at time 141.364 for reason 'discarding message because the queue is full'
[component_container_isolated-6] [WARN] [1674612886.279423396] [amcl]: AMCL cannot publish a pose or update the transform. Please set the initial pose...
```

closed 2025-11-20 by ahcorde