ros2/rviz #521: Error in rviz2 while sending navigation 2d goal in ros2

pankhurivanjani, 2020-03-20:
Hello! 

I am working on navigation2 using ros2 dashing. While sending nav2 goal, robot isn't responding. Checking the logs lead me to this rviz2 error 

```
[rviz2-3] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 325.41500 but the latest data is at time 325.38000, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 325.61500 but the latest data is at time 325.58400, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 325.81500 but the latest data is at time 325.78800, when looking up transform from frame [base_scan] to frame [map]
[rviz2-3] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 326.01500 but the latest data is at time 325.99200, when looking up transform from frame [base_scan] to frame [map]

```

I tried looking into similar issues with ros1 in other threads, some of which suggested changing queue size in laserscan, I tried that too but it didn't help.

I have done few changes from regular configuration:

1. Enabled static layer in global costmap 

Does it affect the navigation goal sending?

Any suggestion in this matter would be helpful.

Thanks and Regards,
Pankhuri

Martin-Idel, 2020-03-23:
This might be fixed via https://github.com/ros2/rviz/pull/519, since until then RViz clock and the clocks used for publishing the navigation goal are not the same, so something might go wrong there - if you could check with master that would be great.

claireyywang, 2020-04-02:
@pankhurivanjani seems like #519 is merged. Mind checking to see if the error is fixed?

pankhurivanjani, 2020-04-02:
Sure, I will check it and provide you with the review.

claireyywang, 2020-04-02:
> Sure, I will check it and provide you with the review.

thanks!

pankhurivanjani, 2020-04-03:
Hello @claireyywang !
I checked it multiple times but error still persists. I did update the ros2 Dashing navigation before running it.

claireyywang, 2020-04-03:
> Hello @claireyywang !
> I checked it multiple times but error still persists. I did update the ros2 Dashing navigation before running it.

By "ros2 dashing" do you mean the `ros2` or `dashing` branch? The fix is merged into `ros2` (with `default` tag).

pankhurivanjani, 2020-04-03:
I have updated ros2. Not going into specific branch, for more information I wanted to convey I am using dashing.

claireyywang, 2020-04-03:
To get the latest #519 fix, you need to update your current `rviz` folder with `ros2` branch upstream. If you are using source installation, you can use the following commands
```
cd [YOUR ROS2 WORKSPACE]/src/ros2/rviz
git checkout ros2
cd [YOUR ROS2 WORKSPACE]
colcon build -packages-up-to rviz
```
You might have to update packages `rviz` depends on, since dashing is almost two releases older than the latest branch.

pankhurivanjani, 2020-04-06:
Hi @claireyywang 
I had binary release and updating it didn't help so I tried updating the source package. While building I got the error in 

```
pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~/ros2_ws$ sudo colcon build
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> fastcdr
Starting >>> foonathan_memory_vendor
Finished <<< foonathan_memory_vendor [1.82s]                  
Starting >>> poco_vendor                                      
Finished <<< fastcdr [1.87s]
Starting >>> fastrtps
Finished <<< poco_vendor [0.35s]                              
Starting >>> cyclonedds
Finished <<< ament_package [2.65s]                            
Starting >>> ament_cmake_core
Finished <<< fastrtps [1.16s]                              
Starting >>> ament_cppcheck
Finished <<< ament_lint [3.09s]
Finished <<< cyclonedds [0.93s]
Starting >>> ament_flake8
Starting >>> tinydir_vendor                                            
Finished <<< ament_cmake_core [2.12s]                                    
Starting >>> ament_cmake_libraries
Finished <<< tinydir_vendor [1.74s]                            
Starting >>> ament_cmake_export_libraries
Finished <<< ament_cppcheck [2.69s]                            
Starting >>> ament_cmake_include_directories
Finished <<< ament_cmake_libraries [1.15s]                    
Starting >>> ament_cmake_python
Finished <<< ament_cmake_export_libraries [1.14s]
Starting >>> ament_cmake_export_dependencies
Finished <<< ament_flake8 [3.02s]                             
Starting >>> ament_cmake_export_definitions
Finished <<< ament_cmake_include_directories [1.04s]
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_export_dependencies [1.05s]                         
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_python [1.16s]                                      
Starting >>> ament_cmake_test
Finished <<< ament_cmake_export_definitions [1.04s]
Starting >>> ament_cmake_version
Finished <<< ament_cmake_export_include_directories [0.97s]
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_export_link_flags [1.05s]
Starting >>> ament_cmake_target_dependencies
Finished <<< ament_cmake_version [1.14s]                                   
Starting >>> ament_pep257
Finished <<< ament_cmake_test [1.63s]                                      
Starting >>> ament_lint_auto
Finished <<< ament_cmake_export_interfaces [1.59s]
Starting >>> gtest_vendor
Finished <<< ament_cmake_target_dependencies [1.30s]
Starting >>> ament_cmake
Finished <<< gtest_vendor [0.35s]                              
Starting >>> ament_cmake_gtest
Finished <<< ament_lint_auto [1.41s]                           
Starting >>> ament_cmake_pytest
Finished <<< ament_cmake [1.10s]                               
Starting >>> uncrustify_vendor
Finished <<< ament_cmake_gtest [1.05s]                         
Starting >>> libyaml_vendor
Finished <<< ament_pep257 [2.51s]                              
Starting >>> ament_copyright
Finished <<< ament_cmake_pytest [1.87s]                                      
Starting >>> ament_index_python
Finished <<< uncrustify_vendor [1.61s]   
Starting >>> ament_uncrustify
Finished <<< libyaml_vendor [2.17s]                                           
[... 100 lines cut ...]
Starting >>> rviz_ogre_vendor
Finished <<< rviz_assimp_vendor [1.28s]                                
Starting >>> qt_gui_py_common
Finished <<< rcutils [5.62s]
Starting >>> rcpputils                                           
Finished <<< qt_gui [1.44s]                                      
Starting >>> rcl_logging_log4cxx
--- stderr: rviz_ogre_vendor                                                 
make[2]: *** [ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-patch] Error 1
make[1]: *** [CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
Aborted  <<< qt_gui_py_common
Aborted  <<< rcl_logging_log4cxx                                             
Aborted  <<< rcpputils                                       

Summary: 84 packages finished [41.7s]
  1 package failed: rviz_ogre_vendor
  3 packages aborted: qt_gui_py_common rcl_logging_log4cxx rcpputils
  1 package had stderr output: rviz_ogre_vendor
  207 packages not processed

```
I could  not find any existing issue for this problem. Could you suggest something?

hidmic, 2020-04-08:
@pankhurivanjani I'd suggest you don't build using `sudo` -- that'll switch your environment for (apparently) no good reason. 

---

About the issues you're seeing. The error logs you're seeing do not seem related to clock mismatch issues. I'll assume you're using `rviz` default send goal tool. Then,

- Check the topic that tool is configured to publish to and update it if necessary.
- `ros2 topic echo` what you publish and make sure the goal message makes sense (coordinates, frames, timestamp).
- Check navigation2 nodes' logs to see if they're getting any message at all.

If everything else fails, `navigation2` offers its own goal tool for `rviz` that you may leverage.

hidmic, 2020-04-08:
One last thing. This kind of questions are more appropriate for [answers.ros.org](http://answers.ros.org/), where the entire community can chime in and provide some guidance on how to proceed. There, all exchanges remain visible for others to learn from them, as opposed to left forgotten in an eventually closed issue.

So I'll close this now, and encourage you to continue with the discussion over there.

closed 2020-04-08 by hidmic