ros2/rviz #522: rviz2 taking more and more CPU/MEM perceptage

ZhilaiShen, 2020-03-24:
Hi all

I am using Ubuntu 16.04, and running everything in 18.04 docker. I write a ros2 c++ plotter(node) to subscribe to messages and publish rviz2 messages(markers: text, arrow, line_strip, bbox; tf2). For all the markers, I setup a lifetime of 2 sec.

The node publishes all rviz2 messages by using a timer_callback function which is called(published) every 1 second even if there are no valid messages coming in. 

I use rosbags to play data for my plotter. However, when I run my plotter, rosbag, and rviz2 in parallel, the CPU/MEM usage for rviz2 will exceed 100% after ~30 min. When I run the c++ node with rosbag along, no such problem happens. 

I am guessing the problem to be that even if I use lifetime for markers, the rviz2 does not delete it correctly, or it just makes it disappear while not actually delete it. Please let me know if there are any thoughts.

Blast545, 2020-04-02:
Hi there @ZhilaiShen, can you please provide further information to reproduce the issue? Like a minimun version of the plotter that causes the problem.

clalancette, 2023-07-14:
No response in a long time, so closing this out.  If you are still having the issue, please feel free to reopen.

closed 2023-07-14 by clalancette