ros2/rviz #524: rviz_assimp_vendor CMake CMP0012 build error

rotu, 2020-04-06:
CMake 3.16.3 on Ubuntu Focal, building from Master

```
--- stderr: rviz_assimp_vendor                                                            
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimpTargets.cmake:54 (if):
  if given arguments:

    "ON"

  An argument named "ON" appears in a conditional statement.  Policy CMP0012
  is not set: if() recognizes numbers and boolean constants.  Run "cmake
  --help-policy CMP0012" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimp-config.cmake:1 (include)
  CMakeLists.txt:55 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
```

Originally (mis)filed as https://github.com/ros-visualization/rviz/issues/1479

jacobperron, 2020-04-09:
This is caused by a bug in assimp 5.0.0 (I believed fixed in assimp 5.0.1). Here it is reported upstream: https://bugs.launchpad.net/ubuntu/+source/assimp/+bug/1869405

I'm not sure there's any way we can quit the warning from RViz while we wait for a fix upstream...

closed 2020-05-02 by wjwwood

dirk-thomas, 2020-06-03:
> This is caused by a bug in assimp 5.0.0 (I believed fixed in assimp 5.0.1).

This is not fixed in 5.0.1 but on `master` of `assimp`. Atm we see the same warning on macOS: https://ci.ros2.org/view/packaging/job/packaging_osx/1870/cmake/

jacobperron, 2020-06-17:
We're also seeing the warning in PR CI builds. E.g. http://build.ros2.org/job/Rpr__rviz__ubuntu_focal_amd64/2/