ros2/rviz #527: Broken build on Clang11 w/ libc++

rotu, 2020-04-16:
I get these build errors with clang11 and libc++:

```
--- stderr: rviz_rendering                                                                                                    
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/src/rviz_rendering/objects/covariance_visual.cpp:31:
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/include/rviz_rendering/objects/covariance_visual.hpp:38:
In file included from /usr/include/eigen3/Eigen/Dense:1:
In file included from /usr/include/eigen3/Eigen/Core:411:
/usr/include/eigen3/Eigen/src/Core/arch/CUDA/Half.h:527:6: error: invalid operands to binary expression ('std::ostream' (aka 'basic_ostream<char>') and 'float')
  os << static_cast<float>(v);
  ~~ ^  ~~~~~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Core/arch/CUDA/Half.h:526:35: note: candidate function not viable: no known conversion from 'float' to 'const Eigen::half' for 2nd argument
EIGEN_ALWAYS_INLINE std::ostream& operator << (std::ostream& os, const half& v) {
                                  ^
/usr/lib/llvm-11/bin/../include/c++/v1/memory:5071:1: note: candidate template ignored: could not match 'shared_ptr<type-parameter-0-2>' against 'float'
operator<<(basic_ostream<_CharT, _Traits>& __os, shared_ptr<_Yp> const& __p);
^
/usr/lib/llvm-11/bin/../include/c++/v1/complex:1442:1: note: candidate template ignored: could not match 'complex<type-parameter-0-0>' against 'float'
operator<<(basic_ostream<_CharT, _Traits>& __os, const complex<_Tp>& __x)
^
/usr/lib/llvm-11/bin/../include/c++/v1/string_view:774:1: note: candidate template ignored: could not match 'basic_string_view<type-parameter-0-0, type-parameter-0-1>' against 'float'
operator<<(basic_ostream<_CharT, _Traits>& __os,
^
/usr/lib/llvm-11/bin/../include/c++/v1/string:4346:1: note: candidate template ignored: could not match 'basic_string<type-parameter-0-0, type-parameter-0-1, type-parameter-0-2>' against 'float'
operator<<(basic_ostream<_CharT, _Traits>& __os,
^
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/src/rviz_rendering/objects/covariance_visual.cpp:31:
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/include/rviz_rendering/objects/covariance_visual.hpp:38:
In file included from /usr/include/eigen3/Eigen/Dense:1:
In file included from /usr/include/eigen3/Eigen/Core:433:
/usr/include/eigen3/Eigen/src/Core/IO.h:142:8: error: no type named 'streamsize' in namespace 'std'
  std::streamsize explicit_precision;
  ~~~~~^
/usr/include/eigen3/Eigen/src/Core/IO.h:163:8: error: no type named 'streamsize' in namespace 'std'
  std::streamsize old_precision = 0;
  ~~~~~^
/usr/include/eigen3/Eigen/src/Core/IO.h:164:43: error: implicit instantiation of undefined template 'std::__1::basic_ostream<char, std::__1::char_traits<char>>'
  if(explicit_precision) old_precision = s.precision(explicit_precision);
                                          ^
/usr/lib/llvm-11/bin/../include/c++/v1/iosfwd:124:32: note: template is declared here
    class _LIBCPP_TEMPLATE_VIS basic_ostream;
                               ^
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/src/rviz_rendering/objects/covariance_visual.cpp:31:
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/include/rviz_rendering/objects/covariance_visual.hpp:38:
In file included from /usr/include/eigen3/Eigen/Dense:1:
In file included from /usr/include/eigen3/Eigen/Core:433:
/usr/include/eigen3/Eigen/src/Core/IO.h:173:27: error: implicit instantiation of undefined template 'std::__1::basic_stringstream<char, std::__1::char_traits<char>, std::__1::allocator<char>>'
        std::stringstream sstr;
                          ^
/usr/lib/llvm-11/bin/../include/c++/v1/iosfwd:139:32: note: template is declared here
    class _LIBCPP_TEMPLATE_VIS basic_stringstream;
                               ^
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/src/rviz_rendering/objects/covariance_visual.cpp:31:
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/include/rviz_rendering/objects/covariance_visual.hpp:38:
In file included from /usr/include/eigen3/Eigen/Dense:1:
In file included from /usr/include/eigen3/Eigen/Core:433:
/usr/include/eigen3/Eigen/src/Core/IO.h:185:16: error: implicit instantiation of undefined template 'std::__1::basic_ostream<char, std::__1::char_traits<char>>'
    if(width) s.width(width);
               ^
/usr/lib/llvm-11/bin/../include/c++/v1/iosfwd:124:32: note: template is declared here
    class _LIBCPP_TEMPLATE_VIS basic_ostream;
                               ^
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/src/rviz_rendering/objects/covariance_visual.cpp:31:
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/include/rviz_rendering/objects/covariance_visual.hpp:38:
In file included from /usr/include/eigen3/Eigen/Dense:1:
In file included from /usr/include/eigen3/Eigen/Core:433:
/usr/include/eigen3/Eigen/src/Core/IO.h:190:19: error: implicit instantiation of undefined template 'std::__1::basic_ostream<char, std::__1::char_traits<char>>'
      if (width) s.width(width);
                  ^
/usr/lib/llvm-11/bin/../include/c++/v1/iosfwd:124:32: note: template is declared here
    class _LIBCPP_TEMPLATE_VIS basic_ostream;
                               ^
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/src/rviz_rendering/objects/covariance_visual.cpp:31:
In file included from /opt/ros/master/src/ros2/rviz/rviz_rendering/include/rviz_rendering/objects/covariance_visual.hpp:38:
In file included from /usr/include/eigen3/Eigen/Dense:1:
In file included from /usr/include/eigen3/Eigen/Core:433:
/usr/include/eigen3/Eigen/src/Core/IO.h:198:27: error: implicit instantiation of undefined template 'std::__1::basic_ostream<char, std::__1::char_traits<char>>'
  if(explicit_precision) s.precision(old_precision);
                          ^
/usr/lib/llvm-11/bin/../include/c++/v1/iosfwd:124:32: note: template is declared here
    class _LIBCPP_TEMPLATE_VIS basic_ostream;
                               ^
8 errors generated.
make[2]: *** [CMakeFiles/rviz_rendering.dir/build.make:310: CMakeFiles/rviz_rendering.dir/src/rviz_rendering/objects/covariance_visual.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:301: CMakeFiles/rviz_rendering.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_rendering	[ Exited with code 2 ]
```

hidmic, 2020-04-23:
Hmm this looks a lot like an upstream issue. What Eigen version are you using?

rotu, 2020-04-27:
I'm using `libeigen3-dev` version 3.3.7-2 installed from apt on Ubuntu Focal.

rotu, 2020-04-27:
I think it's resolved by this https://gitlab.com/libeigen/eigen/-/merge_requests/65
Eigen assumes `<sstream>` has been included already.

I'm not sure if we should fix it here, wait for the next upstream release, or vendor/patch the Eigen3 package.

hidmic, 2020-04-27:
Hmm, brittle as it is, patching here may be the simplest path forward. @ros2/team thoughts?

clalancette, 2020-04-27:
> Hmm, brittle as it is, patching here may be the simplest path forward. @ros2/team thoughts?

Yeah, I'm on board with that, with a comment and a link to where it is fixed upstream.

wjwwood, 2020-04-28:
👍 from me too, to fix it here and link to the upstream issue.

PR #546 (2020-05-14): Work around Eigen build bug
Fixes #527

Note that this PR is a rebase of #530 onto the latest so we can go for green CI.  Assuming we get that, I'm going to go ahead and merge based on the approval in #530.  @rotu FYI.

CI:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
commit 0e274fa2 (2020-05-14): Work around Eigen build bug (#546)

Fixes #527
closed 2020-05-14 by clalancette