ros2/rviz #535: Build fail on Ubuntu 18.04

evshary, 2020-05-04:
I built ROS 2 foxy (master branch of [ros2/ros2](https://github.com/ros2/ros2/blob/master/ros2.repos)) from source code on Ubuntu 18.04.
(With release build: `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`)
However it failed while building package rviz_assimp_vendor. The error message is like below
```
--- stderr: rviz_assimp_vendor                                                                                                                                                                             
CMake Error at CMakeLists.txt:81 (message):
  failed to find assimpTargets.cmake as expected

---
Failed   <<< rviz_assimp_vendor	[ Exited with code 1 ]
```

I guess the error is related to [here](https://github.com/ros2/rviz/blob/ros2/rviz_assimp_vendor/CMakeLists.txt#L78). Ubuntu 18.04 can't find this file, so it generates error.
Although foxy only supports Ubuntu 20.04, maybe the path should not be fixed?

clalancette, 2020-05-04:
See https://github.com/ros2/rviz/pull/536, which fixes the issue.

wjwwood, 2020-05-04:
But also, you shouldn't expect master to compile on 18.04 moving forward. We'll try to keep that the case, but will not avoid a breaking change if we need to do it. The Eloquent branch will continue to be the most stable thing to build on bionic.

PR #536 (2020-05-04): Make rviz_assimp_vendor hack specific to Ubuntu Focal.

This will keep it working on Ubuntu Bionic.

Fixes #535
closed 2020-05-04 by wjwwood