ros2/rviz #547: Segfault when having several topicproperties in a plugin and clicking second one

Tamme, 2020-05-20:
When creating a custom display plugin with several ros topic properties. The first works okay, but when clicking on second one to open the dropdown list, segfault happens in 
``` 
rviz_common::properties::RosTopicProperty::fillTopicList() () from /opt/ros/eloquent/lib/librviz_common.so
```

Stacktrace:
```
#0  0x00007ffff7afe02d in rviz_common::properties::RosTopicProperty::fillTopicList() () at /opt/ros/eloquent/lib/librviz_common.so
#1  0x00007ffff6d24555 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff7b615ef in rviz_common::properties::EditableEnumProperty::requestOptions(rviz_common::properties::EditableEnumProperty*) ()
    at /opt/ros/eloquent/lib/librviz_common.so
#3  0x00007ffff7af5228 in rviz_common::properties::EditableEnumProperty::createEditor(QWidget*, QStyleOptionViewItem const&) ()
    at /opt/ros/eloquent/lib/librviz_common.so
#4  0x00007ffff7af7e0f in rviz_common::properties::PropertyTreeDelegate::createEditor(QWidget*, QStyleOptionViewItem const&, QModelIndex const&) const () at /opt/ros/eloquent/lib/librviz_common.so
#5  0x00007ffff7559850 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff7559a58 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#7  0x00007ffff7559c7a in QAbstractItemView::edit(QModelIndex const&, QAbstractItemView::EditTrigger, QEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#8  0x00007ffff755b4bb in QAbstractItemView::currentChanged(QModelIndex const&, QModelIndex const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff75c7936 in QTreeView::currentChanged(QModelIndex const&, QModelIndex const&) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff7afacb1 in rviz_common::properties::PropertyTreeWidget::currentChanged(QModelIndex const&, QModelIndex const&) ()
    at /opt/ros/eloquent/lib/librviz_common.so
#11 0x00007ffff75569a1 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff6d24555 in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff6ca168a in QItemSelectionModel::currentChanged(QModelIndex const&, QModelIndex const&) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff6ca197e in QItemSelectionModel::setCurrentIndex(QModelIndex const&, QFlags<QItemSelectionModel::SelectionFlag>) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
```
Code I'm using:
```
topic_property1_ = new RosTopicProperty("Topic1", "", "", "", this, SLOT(updateTopic()));

topic_property2_ = new RosTopicProperty("Topic2", "", "", "", this, SLOT(updateTopic()));
```
I'm using ROS Eloquent on Ubuntu 18.04. It worked fine in ROS1 Melodic.

hidmic, 2020-05-21:
@Tamme AFAIK, going from ROS 1 to ROS 2, `rviz` has changed. And without means of reproducing this, unless you can prove this is a fundamental issue with `rviz`, I'd be inclined to close this as a likely non-issue and encourage you to make this question (and any other question of this kind) in `answer.ros.org`, where you can get guidance while debugging your custom plugin. Can you provide any other information?

Tamme, 2020-05-22:
Yes. That was my bad. Hadn't initialized the variables correctly.

closed 2020-05-22 by Tamme