ros2/rviz #549: Eloquent branch not building Ubuntu 20.04

mganglb, 2020-05-28:
Hello, 

i tryed to build Eloquent from source on Ubuntu 20.04 and i got the following error.

`colcon build --symlink-install --packages-select rviz_ogre_vendor`

```
Starting >>> rviz_ogre_vendor
--- stderr: rviz_ogre_vendor                              
CMake Error at Components/Terrain/cmake_install.cmake:95 (file):
  file RPATH_CHANGE could not write new RPATH:

    /home/*/ros2_eloquent/build/rviz_ogre_vendor/ogre_install/lib

  to the file:

    /home/*/ros2_eloquent/build/rviz_ogre_vendor/ogre_install/lib/libOgreTerrain.so.1.12.1

  No valid ELF RPATH or RUNPATH entry exists in the file; Error reading ELF
  identification.
Call Stack (most recent call first):
  Components/cmake_install.cmake:45 (include)
  cmake_install.cmake:181 (include)

make[3]: *** [Makefile:108: install] Error 1
make[2]: *** [CMakeFiles/ogre-v1.12.1.dir/build.make:74: ogre-v1.12.1-prefix/src/ogre-v1.12.1-stamp/ogre-v1.12.1-install] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ogre-v1.12.1.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
```

Martin-Idel, 2020-05-30:
Can you build the current master? Eloquent is not targeted to Ubuntu 20.04 and foxy is right around the corner (June 5th), so I'd propose to just use foxy on Ubuntu 20.04. If you have problems with the current master, that's a bigger problem!

jacobperron, 2020-06-01:
@Martin-Idel is correct; Eloquent does not support Ubuntu 20.04. Please try the master branch.

wjwwood, 2020-06-02:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2020-06-02 by wjwwood

mganglb, 2020-06-03:
Hello, ty for your quick response. 
I know that foxy is around the corner and that eloquent is not targeting Ubuntu 20.04, but i was curious.

But I found out that the following commit of Eloquent works on Ubuntu 20.04. Maybe this is useful for someone else.
```
  ros2/rviz:
    type: git
    url: https://github.com/ros2/rviz.git
    version: 592a0c20738d71c749c768530a9fe0fc2ef4a6f8
```