ros2/rviz #552: cannot change pointcloud size when rviz2 is auto-sized to fit screen

flynneva, 2020-06-01:
[similar to this discussion over on rviz for Noetic](https://github.com/ros-visualization/rviz/issues/1508)....when I run rviz2 on my machine it works great until I auto-expand it to fit the screen (either half the screen or full screen). once this is done I cannot change the size of the pointcloud topic I am displaying.

**Full screen auto-sized (cannot change size of points):**
[image]

**non-auto sized (can change size of points):**
[image]

Tested on:
- Intel x86
- Ubuntu 20.04
- Foxy built from source

```
[INFO] [1591020425.764452318] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1591020425.764746767] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1591020425.817653351] [rviz2]: Stereo is NOT SUPPORTED
```

wjwwood, 2020-06-02:
Thanks, we'll keep an eye on that other issue (https://github.com/ros-visualization/rviz/issues/1508) and investigate here if needed, but I assume it's the same issue (likely with mesa).

Martin-Idel, 2020-06-07:
Since it came up in the other thread: RViz 2 uses Ogre 1.12.1 out of the box, so I guess recent Ogre won't help. If you want to give it a try, I could update to 1.12.6 and push a branch for you to try out (I have neither upgraded to 20.04 nor do I have an Intel x86 at the moment).

flynneva, 2020-06-13:
more than happy to help with testing

clalancette, 2023-07-14:
Based on the conversation in https://github.com/ros-visualization/rviz/issues/1508, this seems to have been a bug in mesa.  And it now seems to be fixed.  Given that, I'm going to close this out.  But feel free to reopen if you still see this issue.

closed 2023-07-14 by clalancette