ros2/rviz #554: Unable to build overlaying a fat binary installation

jacobperron, 2020-06-01:
To reproduce, install Foxy from a fat archive:

- [Linux](https://ci.ros2.org/view/packaging/job/packaging_linux/)
- [macOS](https://ci.ros2.org/view/packaging/job/packaging_osx/)
- [Windows](https://ci.ros2.org/view/packaging/job/packaging_windows/)

Then, try to build `rviz` in an overlay:

```
$ colcon build --packages-up-to rviz2
```

We get CMake errors at `rviz_rendering`:

<details>
<summary>Error:</summary>
<pre>
<code>
--- stderr: rviz_rendering
CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in CMakeLists.txt:
  Imported target "resource_retriever::resource_retriever" includes
  non-existent path

    "C:/ci/ws/install/opt/libcurl_vendor/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
</code>
</pre>
</details>

I've only tried on Windows, but I presume we'll see the same issue on other platforms.

cottsay, 2020-06-02:
I stumbled across this quite a while ago during development of the CI jobs on build.ros2.org. I filed this issue at that time: https://github.com/ros2/ros2/issues/606

dirk-thomas, 2020-06-02:
> I've only tried on Windows, but I presume we'll see the same issue on other platforms.

I tried it on Linux and the overlay workspace containing this repo build just fine for me.

Checking the content of the Windows archive it looks like that it *does* contain the directory `opt/libcurl_vendor/include`?

jacobperron, 2020-06-02:
> Checking the content of the Windows archive it looks like that it does contain the directory opt/libcurl_vendor/include?

Doing a search in the archive, I see that the prefix `C:/ci/ws/install` is appended to the include and lib paths. This prefix does not exist on my host and is an artifact of the CI host.

```
share\resource_retriever\cmake\resource_retrieverExport.cmake:  INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;C:/ci/ws/install/opt/libcurl_vendor/include;${_IMPORT_PREFIX}/include"
share\resource_retriever\cmake\resource_retrieverExport.cmake:  INTERFACE_LINK_LIBRARIES "ament_index_cpp::ament_index_cpp;C:/ci/ws/install/opt/libcurl_vendor/lib/libcurl_imp.lib"
```

clalancette, 2020-06-02:
All right, so I tried this out on all 3 of our platforms.  This particular issue does not happen for me on Linux or macOS, but it does happen on Windows.

There is a separate, but related, problem that happens on all platforms.  When trying to build a downstream rviz_plugin against the Linux packaging build, we get a different set of errors:

```
CMake Error in CMakeLists.txt:
  Imported target "rviz_common::rviz_common" includes non-existent path

    "/home/jenkins-agent/workspace/packaging_linux/ws/install/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

I'll open a separate bug to track that, since it does seem different than this one.

dirk-thomas, 2020-06-03:
Comparing the content of the fat archives on Linux and Windows shows that:

* On Windows `resource_retrieverExport.cmake` contains absolute paths to `libcurl_vendor` in the install prefix.
* On Linux it references `/usr/(lib|include)/x86_64-linux-gnu` which is still absolute but not tied to the install prefix.

So obviously the `resource_retriever` package isn't relocatable on Windows. ros/resource_retriever#46 resolves that part.

---

Similar problems are addressed by ros2/geometry2#271.

jacobperron, 2020-06-03:
With the referenced PRs, it looks like this issue is resolved 👍

dirk-thomas, 2020-06-04:
> With the referenced PRs, it looks like this issue is resolved 👍 

Thanks for confirming!

closed 2020-06-04 by dirk-thomas