ros2/rviz #557: [laser scan] segmentation on invalid topic

Karsten1987, 2020-06-02:
The laser scan plugin makes RViz crash [segmentation fault -11] when selecting a wrong topic.

Steps to reproduce:

* ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.xml
* Start Rviz2, add laser scan plugin, select "goal_pose" as topic.

Karsten1987, 2020-06-11:
the bt I get from clang:
```
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff735d72c6 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff73692bf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff735416a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff7071d641 libc++abi.dylib`abort_message + 231
    frame #4: 0x00007fff7071d7c7 libc++abi.dylib`default_terminate_handler() + 243
    frame #5: 0x00007fff71cd0eeb libobjc.A.dylib`_objc_terminate() + 105
    frame #6: 0x00007fff7072919e libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #7: 0x00007fff70729213 libc++abi.dylib`std::terminate() + 51
    frame #8: 0x0000000148b62cef librviz_default_plugins.dylib`__clang_call_terminate + 15
    frame #9: 0x0000000148d3218c librviz_default_plugins.dylib`tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> >, rviz_common::transformation::FrameTransformer>::~MessageFilter(this=0x0000000152e6b730) at message_filter.h:260:3
    frame #10: 0x0000000148d30795 librviz_default_plugins.dylib`tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> >, rviz_common::transformation::FrameTransformer>::~MessageFilter(this=0x0000000152e6b730) at message_filter.h:249:3
    frame #11: 0x0000000148d2fa39 librviz_default_plugins.dylib`std::__1::__shared_ptr_emplace<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> >, rviz_common::transformation::FrameTransformer>, std::__1::allocator<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> >, rviz_common::transformation::FrameTransformer> > >::__on_zero_shared(this=0x0000000152e6b710) at memory:3706:23
    frame #12: 0x0000000148b68f7a librviz_default_plugins.dylib`std::__1::__shared_count::__release_shared(this=0x0000000152e6b710) at memory:3540:9
    frame #13: 0x0000000148b68f1f librviz_default_plugins.dylib`std::__1::__shared_weak_count::__release_shared(this=0x0000000152e6b710) at memory:3582:27
    frame #14: 0x0000000148b68eec librviz_default_plugins.dylib`std::__1::shared_ptr<rclcpp::Subscription<sensor_msgs::msg::FluidPressure_<std::__1::allocator<void> >, std::__1::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<sensor_msgs::msg::FluidPressure_<std::__1::allocator<void> >, std::__1::allocator<void> > > >::~shared_ptr(this=0x00007ffeefbfb948) at memory:4518:19
    frame #15: 0x0000000148b668f5 librviz_default_plugins.dylib`std::__1::shared_ptr<message_filters::Subscriber<sensor_msgs::msg::FluidPressure_<std::__1::allocator<void> > > >::~shared_ptr(this=0x00007ffeefbfb948) at memory:4516:1
    frame #16: 0x0000000148d47203 librviz_default_plugins.dylib`std::__1::shared_ptr<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> >, rviz_common::transformation::FrameTransformer> >::reset(this=0x000000010939ec90) at memory:4653:5
    frame #17: 0x0000000148cea127 librviz_default_plugins.dylib`rviz_common::MessageFilterDisplay<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> > >::unsubscribe(this=0x000000010939eb60) at message_filter_display.hpp:155:16
    frame #18: 0x0000000148cec57d librviz_default_plugins.dylib`rviz_common::MessageFilterDisplay<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> > >::resetSubscription(this=0x000000010939eb60) at message_filter_display.hpp:146:5
    frame #19: 0x0000000148ce98c5 librviz_default_plugins.dylib`rviz_common::MessageFilterDisplay<sensor_msgs::msg::LaserScan_<std::__1::allocator<void> > >::transformerChangedCallback(this=0x000000010939eb60) at message_filter_display.hpp:141:5
    frame #20: 0x000000010015ca04 librviz_common.dylib`rviz_common::_RosTopicDisplay::qt_static_metacall(_o=0x000000010939eb60, _c=InvokeMetaMethod, _id=1, _a=0x00007ffeefbfbaa8) at moc_ros_topic_display.cpp:77:21
    frame #21: 0x000000010420948c QtCore`___lldb_unnamed_symbol2573$$QtCore + 1468
    frame #22: 0x0000000100154c85 librviz_common.dylib`rviz_common::properties::Property::changed(this=0x0000000109391440) at moc_property.cpp:174:5
    frame #23: 0x0000000100219c7f librviz_common.dylib`rviz_common::properties::Property::setValue(this=0x0000000109391440, new_value=0x00007ffeefbfbb80) at property.cpp:137:12
    frame #24: 0x000000010020e56b librviz_common.dylib`rviz_common::properties::EditableEnumProperty::setString(this=0x0000000109391440, str=0x00007ffeefbfbcd8) at editable_enum_property.cpp:95:3
    frame #25: 0x00000001001534d3 librviz_common.dylib`rviz_common::properties::EditableEnumProperty::qt_static_metacall(_o=0x0000000109391440, _c=InvokeMetaMethod, _id=1, _a=0x00007ffeefbfbce0) at moc_editable_enum_property.cpp:86:21
  * frame #26: 0x000000010420948c QtCore`___lldb_unnamed_symbol2573$$QtCore + 1468
```

sloretz, 2020-06-25:
@Karsten1987 friendly ping :)

Is this only on OSX? Does it happen in only in Foxy, or only with the latest sources?

Karsten1987, 2020-06-25:
I don't have a ubuntu setup with X, so I can't tell if it's happening on other platforms as well.

SebastianGrans, 2020-06-26:
This is related to #568.

spiralray, 2020-06-26:
I found [a interesting article](https://qiita.com/Nabetani/items/498ae03023553087ff31) written in Japanese.

I summarized it below.

## When does this error occur?
```
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
```
This error is occurred if **the order of the constructor is incorrect**.
The constructor of `std::thread` must be called **after** `std:: mutex `.

## Why this problem doesn't occur in Ubuntu?
This problem only occur with Clang.
In Ubuntu, g++ is used as the default compiler, so this problem is not brought to the fore.

# so, what should we do?
We should check the order of constructors and destructors of `std::mutex` and `std::thread`.

spiralray, 2020-07-04:
I found a part of the cause.

[image]

In my opinion, `MessageFilterDisplay::unsubscribe` should be as follows.

```cpp
  virtual void unsubscribe()
  {
    if (subscription_) {
      subscription_.reset();
    }
    if (tf_filter_) {
      tf_filter_.reset();
    }
  }
```

spiralray, 2020-07-04:
I found and fixed additional failure.

- tf_filter_ should not have nullptr
- The order of destructor maybe not good.

```cpp
class MessageFilterDisplay : public _RosTopicDisplay
{
public:
  MessageFilterDisplay()
  : subscription_(),
    tf_filter_(),
    messages_received_(0)
  {
    ...
  }

  virtual void unsubscribe()
  {
    if (subscription_) {
      subscription_.reset();
    }
    if (tf_filter_) {
      tf_filter_.reset();
    }
  }
```

closed 2020-07-17 by Karsten1987