ros2/rviz #568: [osx] Cannnot start-up rviz2 if Path is registered on display list

spiralray, 2020-06-22:
# Overview
If **rviz_default_plugins/Path** is registered on the list of display, rviz2 cannot start up on my Mac OS X.

```shell
% ros2 run rviz2 rviz2
[INFO] [1592832959.881299381] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1592832959.881398239] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [1592832959.910476767] [rviz2]: Stereo is NOT SUPPORTED
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
```

After I delete setting files `~/rviz2/`, it runs again.

Has anyone run into the same problem?

## System
target platform: Max OS X Catalina (sry, I don't have Mojave)
ROS2 version: Dashing, Eloquent and Foxy

# Backtrace
```
% ros2 run --prefix 'lldb' rviz2 rviz2
(lldb) target create "/Users/spiralray/ros2_foxy/ros2-osx/lib/rviz2/rviz2"
Current executable set to '/Users/spiralray/ros2_foxy/ros2-osx/lib/rviz2/rviz2' (x86_64).
(lldb) r
Process 19980 launched: '/Users/spiralray/ros2_foxy/ros2-osx/lib/rviz2/rviz2' (x86_64)
1592832387.175330 [0]    7351006: using network interface en0 (udp/192.168.11.30) selected arbitrarily from: en0, en1
2020-06-22 22:26:27.620254+0900 rviz2[19980:7351132] flock failed to lock maps file: errno = 35
2020-06-22 22:26:27.620891+0900 rviz2[19980:7351132] flock failed to lock maps file: errno = 35
[INFO] [1592832387.878310810] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1592832387.878406750] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [1592832387.914530571] [rviz2]: Stereo is NOT SUPPORTED
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Process 19980 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff68f1633a libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff68f1633a <+10>: jae    0x7fff68f16344            ; <+20>
    0x7fff68f1633c <+12>: movq   %rax, %rdi
    0x7fff68f1633f <+15>: jmp    0x7fff68f10629            ; cerror_nocancel
    0x7fff68f16344 <+20>: retq
Target 0: (rviz2) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff68f1633a libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff68fd2e60 libsystem_pthread.dylib`pthread_kill + 430
    frame #2: 0x00007fff68e9d808 libsystem_c.dylib`abort + 120
    frame #3: 0x00007fff660fd458 libc++abi.dylib`abort_message + 231
    frame #4: 0x00007fff660ee8a7 libc++abi.dylib`demangling_terminate_handler() + 238
    frame #5: 0x00007fff67c295b1 libobjc.A.dylib`_objc_terminate() + 104
    frame #6: 0x00007fff660fc887 libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #7: 0x00007fff660fc829 libc++abi.dylib`std::terminate() + 41
    frame #8: 0x0000000155a2337b librviz_default_plugins.dylib`__clang_call_terminate + 11
    frame #9: 0x0000000155aa7cbf librviz_default_plugins.dylib`tf2_ros::MessageFilter<nav_msgs::msg::Path_<std::__1::allocator<void> >, rviz_common::transformation::FrameTransformer>::~MessageFilter() + 1743
    frame #10: 0x0000000155a23359 librviz_default_plugins.dylib`rviz_common::MessageFilterDisplay<sensor_msgs::msg::FluidPressure_<std::__1::allocator<void> > >::unsubscribe() + 121
    frame #11: 0x0000000155a22532 librviz_default_plugins.dylib`rviz_common::MessageFilterDisplay<sensor_msgs::msg::Image_<std::__1::allocator<void> > >::transformerChangedCallback() + 18
    frame #12: 0x00000001017d32ef QtCore`___lldb_unnamed_symbol2579$$QtCore + 1503
    frame #13: 0x000000010014fb37 librviz_common.dylib`rviz_common::transformation::TransformationManager::transformerChanged(std::__1::shared_ptr<rviz_common::transformation::FrameTransformer>) + 55
    frame #14: 0x00000001001c980f librviz_common.dylib`rviz_common::transformation::TransformationManager::setTransformer(rviz_common::PluginInfo const&) + 319
    frame #15: 0x00000001001c9a19 librviz_common.dylib`rviz_common::transformation::TransformationManager::load(rviz_common::Config const&) + 217
    frame #16: 0x00000001001f91e4 librviz_common.dylib`rviz_common::VisualizationManager::load(rviz_common::Config const&) + 660
    frame #17: 0x00000001001f147b librviz_common.dylib`rviz_common::VisualizationFrame::load(rviz_common::Config const&) + 75
    frame #18: 0x00000001001ed192 librviz_common.dylib`rviz_common::VisualizationFrame::loadDisplayConfig(QString const&) + 1234
    frame #19: 0x00000001001ea0c1 librviz_common.dylib`rviz_common::VisualizationFrame::initialize(std::__1::weak_ptr<rviz_common::ros_integration::RosNodeAbstractionIface>, QString const&) + 2113
    frame #20: 0x00000001001fac21 librviz_common.dylib`rviz_common::VisualizerApp::init(int, char**) + 2945
    frame #21: 0x000000010001d4d4 rviz2`main + 1140
    frame #22: 0x00007fff68dcecc9 libdyld.dylib`start + 1
```

SebastianGrans, 2020-06-26:
I too have this problem.

spiralray, 2020-07-04:
This issue is duplicated with 557, so I close this. sry

closed 2020-07-04 by spiralray