ros2/rviz #571: rviz2 conflict with ros1

MakinoharaShouko, 2020-06-27:
I am working under a work space in which ros-eloquent and ros-melodic exist at the same time and when I launch rviz2, I get the following error
```
[rospack] Error: package 'rviz_common' not found
[librospack]: error while executing command
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[rospack] Error: package 'rviz_common' not found
[librospack]: error while executing command
[rospack] Error: package 'rviz_common' not found
[librospack]: error while executing command
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[rospack] Error: package 'rviz_default_plugins' not found
[librospack]: error while executing command
[rospack] Error: package 'rviz_default_plugins' not found
[librospack]: error while executing command
[rospack] Error: package 'rviz_common' not found
[librospack]: error while executing command
[rospack] Error: package 'rviz_common' not found
[librospack]: error while executing command
[rospack] Error: package 'rviz_common' not found
[librospack]: error while executing command
/home/makinohara/rviz2_ws/install/lib/rviz2/rviz2: symbol lookup error: /home/makinohara/rviz2_ws/install/lib/librviz_common.so: undefined symbol: _ZNK12class_loader23MultiLibraryClassLoader27getAllAvailableClassLoadersEv
```
When I remove ros-melodic, rviz2 can work properly. I think there might be some conflict between rviz2 and ros1 but I have no idea how to resolve it. The problem persists after building ros2/rviz from source and sourcing `~/rviz2_ws/install/setup.bash`

Martin-Idel, 2020-06-28:
Yes, the issue is most probably that `pluginlib` and `class_loader` (used by `pluginlib`) exist in both ROS and ROS 2. We had similar issues when writing a plugin to play `rosbag`s from ROS with `rosbag2` (see https://github.com/ros2/rosbag2_bag_v2)

Since you said you can compile `rviz2`, this is not a compile-time problem, so I guess the library gets linked correctly at compile-time but the wrong library is loaded at runtime (you can see this by looking at CMake output from building `rviz_common`: Where does it find the `pluginlib`/`class_loader`? Compare this to the output of `lddtree` against your `rviz2` library).

Now for runtime, it might be possible to solve the issue by reordering the `LD_LIBRARY_PATH` - make sure that ROS 2 libraries are loaded before ROS libraries (if you source both in your `.bashrc` try to switch the order). `lddtree` (via `sudo apt install pax-utils`) can be of help seeing where libraries get resolved.

In any case, it may be that this issue is not completely resolvable as ROS doesn't really bother to correctly version their libraries as far as I know.

MakinoharaShouko, 2020-06-28:
> Yes, the issue is most probably that `pluginlib` and `class_loader` (used by `pluginlib`) exist in both ROS and ROS 2. We had similar issues when writing a plugin to play `rosbag`s from ROS with [...]

I used `lddtree` to check and found that `libclass_loader.so` got changed from `/opt/ros/eloquent/lib/libclass_loader.so` to `/opt/ros/melodic/lib/libclass_loader.so` after sourcing a ros1 package. I have already switched the order of sourcing files by putting the ros1 package at the end of `.bashrc` but the problem still exists

Martin-Idel, 2020-06-29:
I guess you should source the ros 1 packages first, not last, because the setup scripts will prepend the corresponding variables. But nevertheless, this will then break ros1 packages using pluginlib. Using different environments (e.g. differently sourced terminals) may help, but I'm unsure. If I remember correctly, only one version of a library with the same SONAME will be loaded at any given time and I believe the `class_loader.so` doesn't define an explicit SONAME (didn't check though, you can do it yourself by lookin at the output uf `objdump -p class_loader.so`).

dirk-thomas, 2020-07-16:
> I am working under a work space in which ros-eloquent and ros-melodic exist at the same time and when I launch rviz2, I get the following error

Why are you trying to use both ROS 1 and ROS 2 at the same time / in the same workspace? That is not recommended / supported - except for very specific cases like the `ros1_bridge` which comes with very specific instructions how to build it (see its [README](https://github.com/ros2/ros1_bridge/blob/master/README.md)).

MakinoharaShouko, 2020-07-17:
I had packages in eloquent which relies on another package in dashing and melodic. So I have been using eloquent and melodic at the same time. After I switched to a new computer it no longer works. But I recently found that the package I have been using in melodic had eloquent support. So I think it is no longer an issue.

closed 2020-07-17 by MakinoharaShouko