ros2/rviz #573: Laser tf conflict with rviz 2 but works fine with rviz 1

Tahir-Rasheed-437, 2020-07-10:
I am new user of ROS2. I am currently using ROS2-dashing. I have a vrep simulation of my Robot with all the topics (/odom, laser scans, etc). I am trying to visualize the scan data in rviz2. The required tf topics are generated in a node through a launch file with parameters use_sim_time true. I can see the stamp of the tf's are well according to the simulation time. But when i try to visualize the laser scan, it gives the following error.

[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 955741.25000 but the latest data is at time 955741.06250, when looking up transform from frame [laser_0] to frame [odom]
and 
Could not transform from [laser_0] to [odom]

I tried ROS1 melodic and use the ROS1 bridge to see if the same error appears in ROS1 rviz(rviz1). Apparently it is working fine in ROS1 rviz without any error and showing the laser data completely fine. I have verified in tf_tree that all my tf's are available. What could be the possible mistake i have been doing? I am failed to understand as its working in rviz but not in rviz2. In the attached picture you can see ROS1 rviz in the left and rviz2 on the right.

    

[image]

Martin-Idel, 2020-07-10:
This could be related to https://github.com/ros2/rviz/issues/359 which only got merged recently (should be shipped with foxy). Do you have any way to try this out?

Tahir-Rasheed-437, 2020-07-15:
oh. so i have to shift to foxy now? I was trying to avoid that as i had some issues in compiling the libsimExtROS2Interface.(https://github.com/CoppeliaRobotics/simExtROS2Interface) that will generate communication of vrep with foxy. 
For the tf issue in rviz2, i am using rviz in ROS1 using ros bridge but surely i can try to see if rviz2 in foxy is working fine or not.

Martin-Idel, 2020-07-15:
To see whether this actually fixes your problem, probably yes.

However, maybe https://github.com/ros2/rviz/pull/551 should be considered for a backport to dashing (and eloquent?) in any case as we have quite a few tickets related to this problem? @jacobperron maybe?

Tahir-Rasheed-437, 2020-07-20:
Yes its fine in foxy. no error is generated from rviz2 in foxy. the message filter is surely a great help. 
Many thanks

closed 2020-07-20 by Tahir-Rasheed-437

jacobperron, 2020-07-20:
> However, maybe #551 should be considered for a backport to dashing (and eloquent?) in any case as we have quite a few tickets related to this problem?

In lieu of a patch for https://github.com/ros2/rviz/issues/548, I think backporting #551 is a reasonable thing to do. I can take a look at making the backports.