ros2/rviz #575: Rviz Crashing with 0x00007ffff210daa9 in ?? () from /lib/x86_64-linux-gnu/libGLdispatch.so.0

Michael-Equi, 2020-07-14:
# Bug

OS: Ubuntu 20.04
Foxy binaries with RViz installed from apt (8.2.0-1focal.20200708.072202)
DDS: Fast-RTPS

## Replication

Running the Rviz file linked below and a seemingly handful of variations of it I get poor reliability when running it with my robot using the navigation2 stack. Sometimes it crashes as often as once every 30 seconds, other times it lasts for several minutes but it always has the same crash error.

You can download my Rviz file from here https://drive.google.com/file/d/1xV7vV0getF1jkD56sIXHFuQKsFF0VC8V/view

## Error 

I am having a hard time localizing any specific element responsible for the crash other than potentially a DDS or bandwidth problem. With Cyclone Eclipse, Rviz will not crash but the simulation also ends up running nearly 10x slower than with Fast-RTPS. Below is the error that I get from gbd whenever it crashes.

```
Start navigation
[INFO] [1594684476.566996148] [rviz2]: Setting estimate pose: Frame:map, Position(-22.0146, 2.69005, 0), Orientation(0, 0, 0.0939719, 0.995575) = Angle: 0.188222
Start navigation
--Type <RET> for more, q to quit, c to continue without paging--RET
Thread 45 "rviz2" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff937fe700 (LWP 237591)]
0x00007ffff210daa9 in ?? () from /lib/x86_64-linux-gnu/libGLdispatch.so.0

```

Michael-Equi, 2020-07-26:
Update: I have tested this further on both Cyclone DDS and eProsima Fast-RTPS which now both show crashing with the same segfault trace back. This hints to me that it is an issue with Rviz and likely due to how much data is being sent for RViz to display. When I include more widgets such as my robots depth sensors the crashing seems to be more frequent.

wjwwood, 2020-08-20:
I would need the backtrace from gdb to know where the crash is happening, the error above just says it happens in OpenGL, but does not say where it originated from.

daisukes, 2020-09-03:
I got very similar crashes these days.
Here is the backtrace. I'm using 8.2.0 source build.

https://gist.github.com/daisukes/217ba1905af6097fd2e16a6e860eb633

wjwwood, 2020-09-22:
The error is coming from the processing of a polygon message, which passes through tf2 and message filters before being consumed with Ogre and eventually calling OpenGL. It's possible it is a bug in the polygon display class or something. I don't know off hand. I personally don't have time to investigate it and reproduce it, but maybe this will help someone else investigate it. Help wanted.

daisukes, 2020-09-23:
It crashes when it calls `clear()` of Ogre ManualObject.
I suspect a thread issue with the messaging thread and others. 
What if while the other thread is calling `clear()`, the PolygonDisplay calls `clear()`?

Phymin, 2020-10-28:
> I got very similar crashes these days.
> Here is the backtrace. I'm using 8.2.0 source build.
> 
> https://gist.github.com/daisukes/217ba1905af6097fd2e16a6e860eb633

@daisukes Could you tell me how to print this backtrace if I run the binary installed rviz in a launch file? Thank you.

daisukes, 2020-10-28:
@Phymin 

Basic commands are here

```
$ ulimit -c unlimited              # enable coredump

# run your program and get core.### file

# if you have multiple core files, you can check which is your target file by using
$ file core.###                       # see the executable path which generated the core

$ gdb <program path> core.###
(gdb) tb                                 # print backtrace
```

You can find a more detailed example here
http://www.brendangregg.com/blog/2016-08-09/gdb-example-ncurses.html

Phymin, 2020-10-28:
@daisukes Got it, thanks again!

Phymin, 2020-10-29:
@daisukes Hi, sorry for bothering you again, after I set `ulimit -c unlimited` and run the rviz, after the rviz crashed, I didnot find any core.* file under the path of `/opt/ros/foxy/lib/rviz2`, does this mean the binary installed rviz cann't generate coredump file or this crash is not a coredump crash? Here is the error message:
```
[ERROR] [rviz2-1]: process has died [pid 170859, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/phymin/elfin_ros2_ws/install/elfin_navigation/share/elfin_navigation/rviz/elfin_navigation2_view.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_eg5lu478'].
```

daisukes, 2020-10-29:
@Phymin I usually find them in the home directory. My `/proc/sys/kernel/core_pattern` file looks like
```
/home/user/core
```

Phymin, 2020-10-29:
@daisukes Hi, finally I got my rviz2 crash backtrace file just as follows, thank you very much.
```
(gdb) bt
#0  0x00007f78fea4caa9 in  ()
    at /lib/x86_64-linux-gnu/libGLdispatch.so.0
#1  0x00007f78ce03b693 in Ogre::GLStateCacheManager::deleteGLBuffer(unsigned int, unsigned int) ()
    at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
#2  0x00007f78ce0209f9 in Ogre::GLHardwareVertexBuffer::~GLHardwareVertexBuffer() ()
    at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
#3  0x00007f78ce020a24 in Ogre::GLHardwareVertexBuffer::~GLHardwareVertexBuffer() ()
    at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.so.1.12.1
#4  0x00007f78ce019dd4 in std::_Sp_counted_ptr<Ogre::GLHardwareVertexBuffer*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() ()
    at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.---Typ----------Type <RET> for more, q to quit, c to continue without paging--c
so.1.12.1
#5  0x0000563c390eb848 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#6  0x00007f7902ea3abd in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#7  0x00007f7902ed28a8 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#8  0x00007f7902ed28c8 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#9  0x00007f7902ed28e8 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#10 0x00007f7902fc7a68 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#11 0x00007f7902fc7a8c in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#12 0x00007f7902fc764b in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#13 0x00007f7902fc6ab1 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#14 0x00007f7902fc5583 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#15 0x00007f7902fc4366 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#16 0x00007f7902fc470e in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#17 0x00007f7902fc39a4 in  () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#18 0x00007f7902fc2670 in Ogre::VertexBufferBinding::unsetAllBindings() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#19 0x00007f7902fc241c in Ogre::VertexBufferBinding::~VertexBufferBinding() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#20 0x00007f7902fb4804 in Ogre::HardwareBufferManagerBase::destroyVertexBufferBindingImpl(Ogre::VertexBufferBinding*) () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#21 0x00007f7902fb470a in Ogre::HardwareBufferManagerBase::destroyVertexBufferBinding(Ogre::VertexBufferBinding*) () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#22 0x00007f79032bfbc8 in Ogre::VertexData::~VertexData() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#23 0x00007f79030135db in Ogre::ManualObject::ManualObjectSection::~ManualObjectSection() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#24 0x00007f7903013666 in Ogre::ManualObject::ManualObjectSection::~ManualObjectSection() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#25 0x00007f790300e21f in Ogre::ManualObject::clear() () at /opt/ros/foxy/opt/rviz_ogre_vendor/lib/libOgreMain.so.1.12.1
#26 0x00007f78cdbca634 in rviz_default_plugins::displays::PathDisplay::processMessage(std::shared_ptr<nav_msgs::msg::Path_<std::allocator<void> > const>) () at /opt/ros/foxy/lib/librviz_default_plugins.so
#27 0x00007f78cda3b745 in rviz_common::MessageFilterDisplay<nav_msgs::msg::Path_<std::allocator<void> > >::incomingMessage(std::shared_ptr<nav_msgs::msg::Path_<std::allocator<void> > const>) () at /opt/ros/foxy/lib/librviz_default_plugins.so
#28 0x00007f78cda3b9be in std::_Function_handler<void (std::shared_ptr<nav_msgs::msg::Path_<std::allocator<void> > const> const&), std::_Bind<void (rviz_common::MessageFilterDisplay<nav_msgs::msg::Path_<std::allocator<void> > >::*(rviz_common::MessageFilterDisplay<nav_msgs::msg::Path_<std::allocator<void> > >*, std::_Placeholder<1>))(std::shared_ptr<nav_msgs::msg::Path_<std::allocator<void> > const>)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<nav_msgs::msg::Path_<std::allocator<void> > const> const&) () at /opt/ros/foxy/lib/librviz_default_plugins.so
#29 0x00007f78cda4bd94 in message_filters::CallbackHelper1T<std::shared_ptr<nav_msgs::msg::Path_<std::allocator<void> > const> const&, nav_msgs::msg::Path_<std::allocator<void> > >::call(message_filters::MessageEvent<nav_msgs::msg::Path_<std::allocator<void> > const> const&, bool) () at /opt/ros/foxy/lib/librviz_default_plugins.so
#30 0x00007f78cda470eb in tf2_ros::MessageFilter<nav_msgs::msg::Path_<std::allocator<void> >, rviz_common::transformation::FrameTransformer>::transformReadyCallback(std::shared_future<geometry_msgs::msg::TransformStamped_<std::allocator<void> > > const&, unsigned long) () at /opt/ros/foxy/lib/librviz_default_plugins.so
#31 0x00007f78cd449430 in  () at /opt/ros/foxy/lib/libtf2_ros.so
#32 0x00007f78e802b4b1 in tf2::BufferCore::testTransformableRequests() () at /opt/ros/foxy/lib/libtf2.so
#33 0x00007f78e802cd34 in tf2::BufferCore::setTransformImpl(tf2::Transform const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std [... cut ...]
#41 0x00007f79039ec3da in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) () at /opt/ros/foxy/lib/librclcpp.so
#42 0x00007f79039f0bdc in rclcpp::executors::SingleThreadedExecutor::spin() () at /opt/ros/foxy/lib/librclcpp.so
#43 0x00007f78cd44fa72 in  () at /opt/ros/foxy/lib/libtf2_ros.so
#44 0x00007f79037efd84 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#45 0x00007f790228e609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#46 0x00007f790362e293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

jediofgever, 2020-11-03:
this issue has made rviz2 unusable for the past month or so 😢

jacobperron, 2020-11-03:
https://github.com/ros2/rviz/pull/620 fixed a race condition with message displays. I'd be interested to know if it resolves this issue. If someone is able to try out the latest `ros2` branch from source that would be great.

iuhilnehc-ynos, 2020-11-03:
I am interested in this issue, but I have a question about it.

I think that https://github.com/ros2/rviz/pull/620 can fix this issue because the operation `manual_object_->clear();` in `PolygonDisplay::reset` and `PolygonDisplay::processMessage` will be called in one QT event thread, but https://github.com/ros2/rviz/pull/620 not taking advantage of the dedicated thread to process message will slow down the performance of Ogre, especially if there is a concrete Display need to process a complicated message.

Does anyone think we should seperate the two operations `rviz_common::MessageFilterDisplay::processMessage` and `VisualizationManager::onUpdate` into two thread?

BTW: If a update about replacing `SingleThreadedExecutor` with `MultipleThreadedExecutor` in `TransformListener::initThread` in the future, there will be a bottle neck for `rviz_common::MessageFilterDisplay::processMessage` in QT event thread to process all concrete Displays.

Phymin, 2020-11-04:
> #620 fixed a race condition with message displays. I'd be interested to know if it resolves this issue. If someone is able to try out the latest `ros2` branch from source that would be great.

@jacobperron Hi, if I want build rviz2 ros2 branch should  I also build ROS2 from source? Because I cannot pass the build process with the ros2 binary installation.

jacobperron, 2020-11-05:
@Phymin You should be able to build against the Rolling distribution. See [Rolling Ridley install instructions](https://index.ros.org/doc/ros2/Installation/Rolling/).

Phymin, 2020-11-06:
After about one and half hour of build and 30 minutes test, the crash problem doesnot happen again, I will keep testing.

jediofgever, 2020-11-11:
> #620 fixed a race condition with message displays. I'd be interested to know if it resolves this issue. If someone is able to try out the latest `ros2` branch from source that would be great.

When we can expect to see effect of #620 in foxy deb release ?

jacobperron, 2020-11-11:
> When we can expect to see effect of #620 in foxy deb release ?

It's tricky, since #620 is not ABI compatible with Foxy. I don't have time to look at making an ABI compatible fix at the moment, so if somebody would like to try I'm happy to review it.

mehmetkillioglu, 2021-01-20:
Hi. I have noticed that the Navigation2 Default Rviz file causes Rviz2 to crash because of a one Polygon display. This is a workaround instead of a fix. This worked for me but I haven't tested it with a different setup. I am using the Cyclone DDS, with Foxy and Navigation2 installed from binaries.

If you remove/disable the Polygon under Controller Group, with the topic name /local_costmap/published_footprint, it does not crash. Although same polygon plugin works fine with the /global_costmap/published_footprint, local costmap polygon does not.

You can find the modified navigation2 rviz view version down in the gist link.

https://gist.github.com/mehmetkillioglu/0db5b62520b40803779725fa8160ce3d

While using the Navigation2 Bringup, you can pass argument parameter rviz_config_file:=/path/to/file after launch command to use custom rviz view for it.

charlielito, 2021-05-06:
Still having same annoying issue :disappointed: I think this should be fixed in the supposedly stable Foxy release.

clalancette, 2023-07-14:
Given that Foxy is now End-of-Life, I'm going to close this out.  If you are still seeing this issue on newer versions, please feel free to open a new ticket with your information.

closed 2023-07-14 by clalancette