ros2/rviz #576: Crashing unter Windows using ROS2 foxy

michael1309, 2020-07-15:
I tried to start rviz2 under windows by using the command
`ros2 run rviz2 rviz2`, but it crashes with the following error message:
`
[ERROR] [1594824079.660194600] [rviz2]: Ogre::InternalErrorException::InternalErrorException: Could not load dynamic library C:\dev\ros2_foxy\opt\rviz_ogre_vendor\bin\RenderSystem_GL_d.  System Error: Das angegebene Modul wurde nicht gefunden.
 in DynLib::load at C:\ci\ws\build\rviz_ogre_vendor\ogre-v1.12.1-prefix\src\ogre-v1.12.1\OgreMain\src\OgreDynLib.cpp (line 117)
`
But   C:\dev\ros2_foxy\opt\rviz_ogre_vendor\bin\RenderSystem_GL_d.dll exists. 

Any ideas?

dirk-thomas, 2020-07-30:
Did you build ROS 2 Foxy from source?
* If yes, in release or debug?
* If not, which exact binaries did you download?

michael1309, 2020-08-01:
* Q: What install instructions did you follow?
  A: I followed the instruction given under https://index.ros.org/doc/ros2/Installation/Foxy/Windows-Install-Binary/

* Q: Did you build ROS 2 Foxy from source?
  A: No
* F: Which exact binaries did you download?
  A: ros2-foxy-20200710-windows-debug-amd64.zip
  (from https://github.com/ros2/ros2/releases)   

I have used the pre-built debug version. This is when this error occurred.  
After that I used the preinstalled release version. With that rviz2 was running.

BTW: I also tried to build ROS2 from source. Due to several crashes during the building process, I aborted this attempt.

ivanpauno, 2020-08-27:
@mjcarroll friendly ping

clalancette, 2023-07-14:
Given that Foxy is now End-of-Life, I'm going to close this.  If you are still seeing issues with the newer releases (Humble, Iron), please feel free to open a new ticket.

closed 2023-07-14 by clalancette